MaslowCNC/GroundControl #518: GroundControl does not load .NC (with capital letters)

MarmadukeMSB, 2017-12-23:
Mastercam posts with file extension .NC (capital letters).  Ground Control won't recognize this as an nc file until I manually change the extension to have lowercase letters.

BarbourSmith, 2017-12-24:
Thanks for making the issue for this so we don't loose track of it. This should be an easy fix 🤞

BarbourSmith, 2017-12-24:
So after looking into it a little bit I realized that we already have a settings option for which file names work, but for some reason it was disabled. It's under "Ground Control Settings" and looks like this:

[image]

I've created PR #519 which re-enables the setting so you can add things to it, and added .NC to the default list of allowed files. Unfortunately, if you already have Ground Control installed and it has already created a settings file it won't be added to your list and you will have to change the setting manually before GC will open those files for you

blurfl, 2017-12-24:
What is the line I would paste into ground control to enable this?

blurfl, 2017-12-24:
This change does not seem to have the desired effect. The list of file extensions seems to be controlled by the filters defined in groundcontrol.kv/LoadDialog/FileChooser line 437

BarbourSmith, 2017-12-24:
I remembered something like. I think that is why the setting is disabled. I will look into it more and try to understand what all is going on. I know I was able to open files with a capital .NC  after adding ".NC" to the list in the Valid File Extension setting.

Maybe it's time to take this issue as a prompting to do a complete re-write of the file opener. Everyone has always hated the Kivy file chooser and we do have another option which is to use the system default file chooser which is accessable through tkinter (https://stackoverflow.com/questions/26607268/simplest-way-to-open-a-file-in-tkinter) which would make opening a file look like every other program. It feels ugly to have both Kivy and Tkinter in the same program so I've been avoiding it, but it seems like it could solve a lot of issues

blurfl, 2017-12-24:
I've been poking around, and if I could figure out how to give loadDialog.py access to data.config.get() could make the present setup work.
 I agree, though - it would be very nice to replace the Kivy filechooser ! Would adding tkinter affect using buildozer for packaging?

BarbourSmith, 2017-12-27:
I was wondering the same thing, but my hunch is that it would get bundled in the same way pyserial does. We'd want to test it tho 😉

BarbourSmith, 2018-01-10:
I believe that this has been fixed with changes to the file chooser 👍 👍

closed 2018-01-10 by BarbourSmith