MaslowCNC/GroundControl #521: Walk Through for Initial Setup?

krkeegan, 2017-12-27:
Would it be wortwhile to have a walk through of initial settings for new users?

If we don't find a groundcontrol.ini file we could:

- Add a note that no file was found (hint that if you have one figure out why it wasn't found)
- Tell user how to identify proper port
- walk through default measurements that are important.  machinewidth mahineheight rotationradius (rotationradius we calculate, but it is much better to be in the ballpark when this starts)

Just a thought.

BarbourSmith, 2017-12-27:
Yes! I like this idea a lot. Like have a tutorial built into GC for "here's what you do first"

krkeegan, 2017-12-27:
Exactly.  It doesn't need to be much.

Pablonoob, 2018-07-25:
New user here.  Would love a tutorial.  I am quite frustrated.  I've been through the calibration three times.  The first time the sled went off the cutting area.  I probably entered values wrong.  The second time, It made the 5 cuts, but they were obviously not symmetrical to each other and only one of the cuts was perhaps vertical, all other cuts were at some noticeable angle from vertical/horizontal.  I though it was a poor temporary sled, so I built a decent permanent sled and tried to calibrate again.  The cuts were in completely different locations after going through the entire calibration process again.  I would expect them to at least be in the ballpark.  I read somewhere we are supposed to keep re-cutting the calibration lines until there is a small margin of error.  What does that mean?  Margin of error how?  The cuts on the top left and top right are not at the same height.  Do I measure the horizontal distance, or measure straight between the cuts when they are at a significant angle from each other?  When I tried to re-cut the calibration lines, my sled wandered over the top of the work area.  So I'm not exactly sure how to re-cut the calibration lines as there doesn't seem to be a way to center the sled before cutting again.  I went ahead and saved out of the calibration and am now at the home screen and can adjust my sled to the center of the physical workspace at least.  But the diagram showing where my sled is marks it way off in the lower right quadrant.  I can not figure out how to tell Ground Control my sled is actually in the exact center of the workpiece.  I'd think there ought to be a way to manually move the sled to the four corners and mark those locations.  And it boggles my mind that the machine doesn't seem to know when it is dragging the sled off the work area.  I would absolutely love an idiot's guide walk-through to setting this thing up and getting it to know it's boundaries.  I'm scared to do anything on it since it doesn't appear to know where the center of the workspace is.  It was centered (for the most part) before the calibration cuts.  But now it is way off.

BarbourSmith, 2018-07-25:
That sounds like something is quite off. I think you are doing everything correctly except that something is not working right.

Photos could help a lot to diagnose what is going on. My first suspect is a loose cable somewhere giving intermittent signal from one of the motors.

What do you see when you run the motors test by pressing **Actions -> Test Motors/Encoders**?

Pablonoob, 2018-07-26:
When I run the motors test, X, Y, and Z all complete successfully.  The sled moves left/right, up/down, and Z moves back and forth.

I tried setting the sled to the center (X=0, Y=0) on the display and then manually adjusted the chain lengths to as close to equal as possible ... about 67" to put the sled in the almost exact center of the work-space.  I'm using Bar's bolt together design (is that you?) and made it symmetrical.  Then I loaded the Calibration Benchmark Test.  It stopped moving half way through the lower-left corner box and tossed an error ... sorry I can't remember the error but I think it was something along the lines of the sled not keeping up with it's position.
Left = 67 1/16", Right = 67" (measured from sprocket to ring pulley)

After clicking stop and getting the carriage to move again, I clicked the home button.  It now thinks home is a bit higher and to the left of the center of my work-space.  My measurements are now:
Left = 63 1/2", Right = 66 7/8"

If I were slipping links, I'd expect the measurements to be shorter.  Somehow it doesn't appear that Ground Control is accurately capturing position information from the motors or the motors aren't accurately sending the information.

Uh ... here's a picture like you requested ... now it seems to be printing rather well after plugging the wire in and it is homing correctly now.  With the test print, the boxes look a tiny bit trapezoidal but I think I can work with that!  Can't believe it was a loose wire harness.  Thanks so much!
Paul
[image]

davidelang, 2018-07-26:
we need the exact error

links can skip going either direction

blurfl, 2018-07-26:
'here's a picture' 
The picture shows the connector not fully plugged in. The side with motor power is good, but the side with the encoder signals would be intermittent. It could fail in a way that will drive the z-axis mechanism into the end stops and damage the orange 'sacrificial link' on the router. (Voice of experience)