MaslowCNC/GroundControl #523: Export $ Commands to a Text Fie

krkeegan, 2017-12-27:
If people were using Grbl senders or something other than GC, they may appreciate a means to export the $$ settings to a gcode file that could be run by that software to set the settings on their machines.  Again not critical

blurfl, 2017-12-27:
A very good idea. Would this be a checkbox choice in GC Advanced settings, or an automatic, created and updated at the same time groundcontrol.ini gets written?

krkeegan, 2017-12-27:
I meant something exported by GC.  So it would be a guide file where each line looked like

$1=2913.5

And do on.  Then using whatever grbl senders exist, just send this to the firmware rather than having to load GC on the sender machine.

I was thinking just a button, click it and it opens the file save menu.

blurfl, 2017-12-27:
That would be fine. Out of interest, does the output of $$ from the firmware look like that list?

krkeegan, 2017-12-27:
Yes, with the exception that it has comments describing each setting..  it
would work to export from the firmware, but for those not using GC directly
connected to the Arduino this might help

davidelang, 2017-12-27:
great to see this.

now, let's make the triangular calibration calculations into a function inside 
the firmware so that everything could be done from a gcode sender
k

blurfl, 2017-12-27:
How would the man/machine interaction work with a calibration routine run through a gcode sender? What prompt and response methods can you count on being available?

krkeegan, 2017-12-27:
It is doable.  It would look a lot like a command prompt or telnet session.

Something like

Sent: $T
Rcvd: Starting triangular kinematics test cut.
Rcvd: Please enter measured distance in MM as $M=
Sent: $M=1905.2
Rcvd: Rotation radius set to 300
Rcvd: ok

It wouldn't be glamorous.  But if you had a headless system, it would be functional.

davidelang, 2017-12-27:
I was thinking more a matter of defining primitives

for example, to set the motor distance, the pseudocode would be:

set left motor to position 0
set right motor to position  0
feed left motor X 'steps'
feed right motor -Y 'steps'
apply power to both motors
set distance between motors to X-Y (with chain pitch/tooth count calculated in)
cut power to both motors
move right motor back to pos 0

with the 'set distance' item being the only new command.

for the triangular calibration you would use normal commands to feed out chain 
from both side, make the cuts, manually measure, then issue a command like:

do triangular calibration, distance from where I am to the other cut should be X 
but is Y

returns "rotation radius set to 300"

no assumptions about what the correct distances are, nothing built into the 
firmware that knows that things should be a given distance, just the minimum 
that must be passed to do the calculation (ideally using it's current position 
and saved motor distance to avoid having to pass a lot of parameters)

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Scott Smith wrote:

> How would the man/machine interaction work with a calibration routine run through a gcode sender? What prompt and response methods can you count on being available?

At the moment, I'm thinking that this would be a 'enter manual g-code line' 
interface.

The way I'm thinking about things, there should not be prompts and responses, 
but rather one-way commands that use the current machine state plus any info 
needed.

We have most of it in place today with the various B commands.

I'm thinking that we need just a couple extra commands

I believe we already have commands in place to set the position of a motor to X 
and to move the motor (absolute movement, relative movement, apply X% power in a 
direction), so I believe we only need a couple extra commands

1. use the current motor position to define the distance between the motors

   Assumes that you zero'd both motors, ran chain out from one motor, hooked it 
over the other motor and pulled the chain tight before executing this command.

2. triangular calibration

   As I tried to specify this, I realized it will need to be two commands

2a. Triangular calibration step 1 "I am at Y=x" (defaulting to 0)

   Causes the firmware to remember the chain lengths for this coordinate

2b. Triangular calibration step 2 "I am at Y=x2"

   The firmware knows where it thinks it is, and what the chain lengths are for 
here. So given the actual position, and the memory from step 2a, it can 
calculate the chain offset and y offset values.

The process of using these would be to feed out chain from both sides to the 
same amount, drive the sled to the first location, issue command 2a. drive the 
sled to a second position (either a pre-marked position or to an expected 
position and then mark it, move the sled, measure it, move the sled back) and 
issue command 2b.

krkeegan, 2017-12-27:
It all depends on how much work we want the user to have to perform:

All of this _can_ be done by hand with everything that is currently in the machine.  There are ways to report the chain length and machine position.  The user just has to do the math on his own to calculate the motor distance or the correction for the rotationradius.

So the question is how much assistance do we want to provide?

@davidelang I understand your motordistance calibration step.  I don't think I understand your concept for how the triangular calibration would work?

BarbourSmith, 2017-12-27:
> If people were using Grbl senders or something other than GC, they may appreciate a means to export the $$ settings to a gcode file that could be run by that software to set the settings on their machines. 

I think this is a BRILLIANT idea. It's simple and lets you use ANY gcode sender program. I agree it's not critical, but the idea is really elegant

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Kevin Robert Keegan wrote:

> It all depends on how much work we want the user to have to perform:

I'm trying to do the minimum that doesn't require the user to do any math

> @davidelang I understand your motordistance calibration step.  I don't think I understand your concept for how the triangular calibration would work?

in the topic 
https://forums.maslowcnc.com/t/calibration-pattern-for-triangular-kinematics/879/61

I go into a way to do calibration for triangular kinematics from a pure math 
point of view, @rjon17469 implements something similar in excel doing some 
brute-force iteration to find the results.

Basically you use the fact that you know:

0. the math is much simpler if you are exactly at the center of the machine
1. the distance between the motors
2. the chain length at one position
3. the chain length at a second position
4. the distance between these two positions

from this you can calculate the 'missing' chain length and the Y offset needed 
to make things work.

It works best if you can use the exact same code for the calibration iterations 
that you do for the actual movement (no chance of the code getting out of sync 
between the two), which makes doing this in firmware ideal.

krkeegan, 2017-12-27:
> I think this is a BRILLIANT idea. It's simple and lets you use ANY gcode sender program. I agree it's not critical, but the idea is really elegant

For the foreseeable future, i don't think it is too much to ask people to calibrate their machines using GC.  They can save their settings as an exported gcode file which they can run using a gcode sender.  

Then if we add any additional settings, necessitating a wipe of the eeprom, they should be able to run this file to re-import all of their old settings and would only need to update the new values (which may be fine as defaults anyways).

I have added the concept of command line calibration to the firmware issues list.