MaslowCNC/GroundControl #529: Using the file open "type in field" crashes GC 1.02 in Linux

GeroBH, 2017-12-30:
Although .ngc is listed, it does not show the files I have in .ngc
Trying to put *.*, or *.nc or *.ngc in the bottom and clicking 'select' crashes GC with following terminal errors:

[INFO   ] [Clipboard   ] Provider: gtk3(['clipboard_dbusklipper'] ignored)
[CRITICAL] [Cutbuffer   ] Unable to find any valuable Cutbuffer provider.
xclip - OSError: [Errno 2] No such file or directory
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/clipboard/clipboard_xclip.py", line 17, in <module>
    p = subprocess.Popen(['xclip', '-version'], stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception

xsel - OSError: [Errno 2] No such file or directory
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/clipboard/clipboard_xsel.py", line 16, in <module>
    p = subprocess.Popen(['xsel'], stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception

[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 738, in <module>
     GroundControlApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 504, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_sdl2.py", line 663, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_sdl2.py", line 405, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 342, in idle
     self.dispatch_input()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 327, in dispatch_input
     post_dispatch_input(*pop(0))
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 293, in post_dispatch_input
     wid.dispatch('on_touch_up', me)
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:7726)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/behaviors/button.py", line 179, in on_touch_up
     self.dispatch('on_release')
   File "kivy/_event.pyx", line 714, in kivy._event.EventDispatcher.dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:7681)
   File "kivy/_event.pyx", line 1225, in kivy._event.EventObservers.dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:13524)
   File "kivy/_event.pyx", line 1109, in kivy._event.EventObservers._dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:12356)
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang/builder.py", line 64, in custom_callback
     exec(__kvlang__.co_value, idmap)
   File "<string>", line 88, in <module>
   File "kivy/_event.pyx", line 714, in kivy._event.EventDispatcher.dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:7681)
   File "kivy/_event.pyx", line 1225, in kivy._event.EventObservers.dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:13524)
   File "kivy/_event.pyx", line 1149, in kivy._event.EventObservers._dispatch (/tmp/pip-build-kYN182/kivy/kivy/_event.c:13050)
   File "/home/whoareyou4/Maslow/fw-test/GroundControl-master/UIElements/viewMenu.py", line 73, in load
     filename = instance.selection[0]
 IndexError: list index out of range`

BarbourSmith, 2018-01-02:
Well that's no good! Let me see what I can figure out.

BarbourSmith, 2018-01-02:
I'm seeing the same behavior on Windows 10

BarbourSmith, 2018-01-03:
I reopened this because the PR did not address the issue of .ngc files not showing up, they are showing up for me now so it might be fixed, but I didn't want to close the issue until we've fixed the issue that  @GeroBH is seeing.

Here's what it looks like for me:

[image]

blurfl, 2018-01-03:
Looks the same on OSX.

GeroBH, 2018-01-03:
The .ngc are showing now. Not sure what was before, as my focus was on the crash. Anyhow, the .ngc files are showing and loading. Still as soon as I click ->Actions -> open file, the terminal reports:
_[INFO   ] [Clipboard   ] Provider: gtk3(['clipboard_dbusklipper'] ignored)
[CRITICAL] [Cutbuffer   ] Unable to find any valuable Cutbuffer provider.
xclip - OSError: [Errno 2] No such file or directory
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/clipboard/clipboard_xclip.py", line 17, in <module>
    p = subprocess.Popen(['xclip', '-version'], stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception

xsel - OSError: [Errno 2] No such file or directory
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/local/lib/python2.7/dist-packages/kivy/core/clipboard/clipboard_xsel.py", line 16, in <module>
    p = subprocess.Popen(['xsel'], stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception_
It still runs though.

blurfl, 2018-01-03:
Does it open in the correct folder?

BarbourSmith, 2018-01-03:
I vaguely remember running into this issue in the distant past with another project. I think the crucial part is

`xsel - OSError: [Errno 2] No such file or directory`

and that the error was caused by some file not existing somewhere. If I am remembering right everything works normally, and it's safe to ignore the warning

blurfl, 2018-01-03:
Is there a runtime switch that would cause the missing file/directory to be named?

GeroBH, 2018-01-03:
Well, don't know what it was but could fix it :-)
Solution:
sudo apt-get install xclip xsel
Found here: (might be of interest for RPi)
https://github.com/kivy/kivy/issues/4764

Can be closed from my side.

PR #536 (2018-01-03): Fix for 529
Disables the text entry box on the file open window which crashes GC if the input text string is not a valid file.

Still missing a fix for .ngc files not showing up in the preview.

Fixes #529
closed 2018-01-03 by BarbourSmith