MaslowCNC/GroundControl #531: calibration problem switching from quad to triangular kinematics

davidelang, 2018-01-02:
in https://forums.maslowcnc.com/t/is-there-a-kinematics-calibration-guide/1702 we just had another case where switching kinematics models doesn't work as long as the settings for the quad kinematics are non-zero.

Those variables should not be used at all inside the triangular calculations, but we've now has numerous cases where they needed to bet set to 0 or the .ini file recreated from scratch before triangular kinematics work

since calibration is mostly a GC item, it's probably a bug in GC, but we should check the firmware just to be sure that the variables aren't used there

madgrizzle, 2018-01-03:
I'm not sure if this is a problem or not, but its something maybe we can investigate.  If the software is in its initial state, then Kinematics is set to Quadrilateral.  When you go through the calibration process, you have the option of selecting the kinematics type.  If you select Triangular, the config gets written with this value so GC starts to use triangular.  But what if (and a big if) this information does not get relayed to the firmware.  I found a closed issue for Kivy [https://github.com/kivy/kivy/issues/982](url) where on_config_changed() isn't called from a button and it looked like it needed to be called manually as a workaround.    When on_config_changed() is called, it sends the updated settings to the firmware, but if its not called, then GC thinks its in triangular mode and the firmware thinks its in quadrilateral mode.   I think this can be tested by reviewing the log file.

BarbourSmith, 2018-01-03:
I was thinking the exact same thing. I added a print statement to the firmware which will tell us every time the kinematics type gets switched so from now on so we can confirm in the log that the command was received and processed correctly

madgrizzle, 2018-01-03:
Can someone point me to where I can figure out how:

`self.data.gcode_queue.put("$7=" + str(kinematicsType))`

gets written out in the log. Does it appear as $7=0 or $7=1 or something else?  Lost, I am.

blurfl, 2018-01-03:
In log.txt, the GC line has 'Sent: at the start.
I used 
> grep '\$7' log.txt 

to find the instances.Not sure how th eWindows 'find' command works 😁

> Sent: $7=2 
$7=2

madgrizzle, 2018-01-03:
Ok, I'm looking at something different then.  I don't see the $7 in my log file so I guess the update command wasn't sent.  I was thinking the "Sent: B12 I1 J8114.0 K10 M6.35 N3.17 P7560.0 R1 S1300 T0 U34 V7 W0 X0.28 Y0 " was a result of the push_settings_to_machine command but must be sent from somewhere else.

blurfl, 2018-01-03:
Are you using v1.00? That one used the 'B12' command. Starting with v1.01/1.02, the $ commands replace B12...

madgrizzle, 2018-01-03:
ARRGHH!!  thanks .. sigh :)