MaslowCNC/GroundControl #539: touchNumberInput.py crashes when TouchNumberInput called - version 1.01/1.02

blurfl, 2018-01-03:
Running v1.00, using the 'GoTo' button or the Z-Axis numeric Input button work as expected. Starting with v1.01, when using 'GoTo' or the Z-Axis numeric input button, GC crashes with the error:

>    File ".../GroundControl-master-1.01/UIElements/touchNumberInput.py", line 17, in __init__
     self.unitsBtn.text = self.data.units
 AttributeError: 'NoneType' object has no attribute 'units'

In v1.01, the 'Dist to Move' works correctly and calls TouchNumberInput with a second parameter:
> UIElements/frontPage.py:354:        self.popupContent = TouchNumberInput(done=self.dismiss_popup, data=self.data)

Currently, TouchNumberInput is called with only one parameter in three places:
 - The 'GoTo' button:
> UIElements/frontPage.py:413:        self.popupContent = TouchNumberInput(done=self.dismiss_gotoLinePopup)
 - The Z-Axis setting:
> UIElements/zAxisPopupContent.py:26:        self.popupContent = TouchNumberInput(done=self.dismiss_popup)
 - Somewhere 😁 in the Calibration process:
> UIElements/measureMachinePopup.py:232:        self.popupContent = TouchNumberInput(done=self.dismiss_popup)

I've confirmed that changing the calls in frontPage.py and zAxisPopupContent.py adding the second parameter 'data=self.data' makes those seem to work correctly. I haven't looked into measureMachinePopup.py, but expect it will need the same treatment.

BarbourSmith, 2018-01-03:
YES! I will make a pull request right away, although you have already done all the hard work for me 😁 

Thanks @blurfl !!! 👍 👍 💯 ❤️

BarbourSmith, 2018-01-03:
It looks like at some point I caught the ones in measureMachinePopup.py and the Z-Axis, #541 catches the one in the GoTo button which was taking down GC when that button was clicked for me too. Great catch @blurfl !

PR #541 (2018-01-03): Fix missing data link in go to line popup
Fixes the missing link to data which was causing GC to crash when entering a line number to jump to

Fixes #539
commit 618a6346 (2018-01-03): Fix missing data link in go to line popup

Fixes the missing link to data which was causing GC to crash when entering a line number to jump to

Fixes #539
closed 2018-01-03 by BarbourSmith