MaslowCNC/GroundControl #543: Show default value next to each setting

BarbourSmith, 2018-01-04:
It would be nice to show a default value next to each setting to make it a little more clear what has changed

blurfl, 2018-01-05:
kivy's settings module limits the 'very easy' solution to editing the "desc" of each option to add something like
"desc": "The horizontal distance between the center of the motor shafts in MM. - default value 1234.5"
That's inelegant as it doesn't automatically capture changes made to the default values later in main.py build_config().

madgrizzle, 2018-01-05:
I'm not a python programmer, so bear with me.  Could you create the default value dictionaries early on and then when creating the JSONs for Kivy settings, use str.format to substitute values from it?  I found this on stackoverflow.  Solution #3 seems like it might work:

[https://stackoverflow.com/questions/36590496/how-to-substitute-value-for-a-variable-in-a-json-in-python](url)

The same dictionaries could then also be used to populate the defaults in config.setdefaults.  This way, there's only one place the defaults are stored.

BarbourSmith, 2018-01-05:
I like that idea a lot 👍

blurfl, 2018-01-05:
@madgrizzle, your link was right to the point. I'm afraid I 'stole' it 😉 to learn a bit more python programming. I'll pull my PR back if you're working on a patch for this issue. 🙂

madgrizzle, 2018-01-05:
Keep going.. I wasn't working on it.. just threw out a suggestion.

⁣Sent from BlueMail ​

BarbourSmith, 2018-01-10:
This is now a thing! Great work @blurfl

closed 2018-01-10 by BarbourSmith