MaslowCNC/GroundControl #549: MoveHere/MarkHere conversion error in Inches mode

blurfl, 2018-01-08:
Platform: OSX, GC v1.02

The MoveHere and MarkHere functions fail to convert from mm to inches when GC is in Inches mode. This can cause the MarkHere mark to be out of the frame in GC, and the sled to try to leave the workarea to reach the MoveHere point. Looking at the values calculated in modernMachine.py, there doesn't appear to be a conversion applied. The units parameter in data.py doesn't get updated by frontPage.py; is there an easy way to ask frontPage.py for the appropriate conversion?

Steps to replicate the issue:
Start from 0,0 im MM mode. 
Place a mark at x=10mm, y=0mm.
Use MoveHere to traverse correctly to this mark.
Re-center to 0,0.
Change to Inches mode.
Use MoveHere to traverse with a target near the 10mm,0mm mark - beware, the sled will move beyond the target to x=10 inches

Printing what look like the pertinent values from modernMenu.py from a 10mm move in MM mode:
touch.pos[0] 982.0
touch.pos[0] 646.0
self.xPos 9.74437335612
self.yPos 0.28295272448
self.parent.pos[0] 883.183236984
self.parent.pos[1]643.130603344
self.parent.scale 10.1409048488
Sending: G0 X9.744 Y0.283  

Printing what look like the pertinent values from modernMenu.py from the same move in Inches mode:
touch.pos[0] 980.0
touch.pos[0] 644.0
self.xPos 9.54715229648
self.yPos 0.0857316648395
self.parent.pos[0] 883.183236984
self.parent.pos[1]643.130603344
self.parent.scale 10.1409048488
Sending: G0 X9.547 Y0.086

BarbourSmith, 2018-01-08:
That is very strange! I use mine in inches mode all the time so I'm surprised I haven't run into this before.

If I jump in and have a go at fixing it will I be stepping on your toes?

blurfl, 2018-01-08:
My.Insey file is in millimeters mode, but I often move things around in inches mode.

commit cc2bee7d (2018-01-11): Make move here/mark here reference machine units, not gcode units

The move here and mark here commands operated on the scale factor of the gcode file. They should be using the same units as the machine becuase they can send information to the machine.

This PR fixes #549
PR #559 (2018-01-11): Make move here/mark here reference machine units, not gcode units
The move here and mark here commands operated on the scale factor of the gcode file. They should be using the same units as the machine becuase they can send information to the machine.

This PR fixes #549
closed 2018-01-11 by BarbourSmith