MaslowCNC/GroundControl #550: Window loads with large error

BarbourSmith, 2018-01-08:
When I launch GC and it connects to the machine a large positional error is shown like this:

[image]

Making the machine move immediately combines the two indicators into one (or just brings them close together the way we normally expect them to be)

krkeegan, 2018-01-08:
Yup, I i did this.

It doesn't have any effect other than being silly to look at.  The error is in the Firmware.  I fixed it in my current work that I haven't finished yet.

thormj, 2018-02-07:
What do the different colors of crosshairs mean?

BarbourSmith, 2018-02-07:
One is the target location, the other is the machines actual position as roughly computed based on the position error in each axis (where the machine probably actually is)