MaslowCNC/GroundControl #557: Add an Alarm Unlock Button

krkeegan, 2018-01-11:
I am working on an enhancement to the firmware that adds an alarm state.  Currently we have two states tripped by different actions:

- Alarm - Tripped by a soft limit error, the sled tried to move outside the defined work area.  This alarm locks the machine until the $X unlock command is received.  At that point the user can move the sled away from the limit and proceed without any loss of position, as long as the sled isn't in the out of bounds.  In my testing, we can stop before we hit the out of bounds, but if a settings change places the sled in the out of bounds, this may be unrecoverable (at least in the current design).

- Critical - This is tripped if the forward kinematics is unable to place the sled anywhere.  This represents a complete loss of position.  It cannot be unlocked and can only be recovered by recalibrating the chain lengths or updating settings such that forward kinematics is able to place the sled somewhere.

So I think we will need a button to at least perform the unlock for the Alarm state. But perhaps the same button can be used to display the Critical state as well.  

**Does anyone have any suggestions for placement and look of this button?**

blurfl, 2018-01-11:
Does the button need to be available/visible all the time? Could it hide until needed?

davidelang, 2018-01-11:
On Wed, 10 Jan 2018, Kevin Robert Keegan wrote:

> - Alarm - Tripped by a soft limit error, the sled tried to move outside the
> defined work area.  This alarm locks the machine until the $X unlock command
> is received.  At that point the user can [...]

unlock should disable the alarm until it goes out of alarm. That way if you are 
outside the boundry, you can do whatever you want until you are within the 
boundry again.

> - Critical - This is tripped if the forward kinematics is unable to place the
> sled anywhere.  This represents a complete loss of position.  It cannot be
> unlocked and can only be recovered by [...]

you need to unlock things to be able to get things into a sane state, but 
possibly instead of allowing nromal G code movements, only allow motor specific 
B codes (different buttons)

krkeegan, 2018-01-11:
It certainly doesn't need to be visible unless it is in an alarm state.  But the other aspects of the UI should still work, so it can't be a popup.

thormj, 2018-02-07:
Maybe have the ohCrapMenu hidden underneath the screenControls button (set size to 0), and if in alarm state, you can reset the size and pop it out (from StackOverflow, it seems like setting a widget's size to 0 is the best way to hide it)?