MaslowCNC/GroundControl #561: Should 'Return to Center' and 'Home' trigger a Z-axis prompt in manual z-axis mode?

blurfl, 2018-01-11:
Presently, if the z-axis is in manual mode, the user is prompted for z-axis changes during a cut, but not if the 'Return to Center' or 'Home' button are used.
 This may be by design, and it's reasonable to expect the user to manage the z-axis. This is just by way of checking whether this is by design or oversight; comments?

@krkeegan , I'm _really_ enjoying your Fake_Servo mode, useful looking at issues like this 💯👍

krkeegan, 2018-01-11:
Hmm, it looks like this is a specific determination made within GC.  The returnToCenter and home functions skip zaxis changes unless a zaxis is present.

I would say yes, we want the zaxis to retract.

@blurfl glad it is helpful.  I agree I bought two additional $13 arduinos so I could test without harming my functioning system.

BarbourSmith, 2018-01-11:
This is a good catch. I vaguely remember deciding it was the right behavior way back at the dawn of time, but now I agree that it seems smart to ask someone to lift the bit.

I vote that the target behavior is to ask to adjust the depth

Great catch!

blurfl, 2018-01-11:
Handled in PR#562

closed 2018-01-11 by blurfl