MaslowCNC/GroundControl #563: Handle files with large numbers of digits after decimal better

BarbourSmith, 2018-01-11:
Some programs (MakerCAM in particular when in metric mode) will create files which have way too many decimal places after the decimal. 

This can be an issue because the lines can be so long that the won't fit in the Arduino's memory causing a buffer overflow. We have a feature to automatically trim the number of decimal places down which can be enabled in settings, however this is not without issues also because if you have a very short arc where the starting and ending points are close together rounding the number can make it so that the points are rounded to the same number which is interpreted as a full circle instead of a tiny arc

We need to at least warn people that the file they are opening might have issues, and that they have the option to truncate the number of decimals in settings.

It might also be worth doing the truncation before rendering the gcode in Ground Control so that any issues could be seen

davidelang, 2018-01-11:
can we detect the 'arc turned into a circle' at truncation time and just skip 
the move entirely?

Along similar lines, if you have a lot of short line segments and the start and 
end are the same after truncation, they should just be skipped and never sent 
to the firmware.

BarbourSmith, 2018-01-11:
Good point

GeroBH, 2018-01-11:
What if we published more to the local version, https://github.com/shapeoko/makercam that has taken care of the digits but sadly never made it online?

BarbourSmith, 2018-01-11:
That is a good point! I played around with having our own branch of makercam for a little bit, but didn't get too far. That way we could improve things.

There's even a tiny version of it running at http://www.maslowcnc.com/makercam

blurfl, 2018-01-11:
Which fork is the maslow version taken from?

BarbourSmith, 2018-01-12:
I am embarrassed to say that I don't remember. If we are serious about this we should really have our own fork of it. The shapeoko fork is off of the creators branch which is here: https://github.com/Jack000/PartKAM

blurfl, 2018-01-12:
The creator's branch has much more info. Wish the solution didn't rely on Flash.

GeroBH, 2018-01-12:
The local copy of partkam.swf I have is from Fri 25 Oct 2013 18:24:41 +03 and it truncates to 4 digits.

blurfl, 2018-01-12:
@GeroBH, could you try the [Maslow version](http://www.maslowcnc.com/makercam)? It would be very handy if that one, too, truncates to 4 digits!

GeroBH, 2018-01-12:
I might have been to fast. With recent firefox updates it might have lost its function. swf on linux... Investigating now.

blurfl, 2018-01-12:
It looks to me like the Maslow version truncates to four places in metric. All the links above seem to truncate to four places in inches. @GeroBH, can you confirm?

GeroBH, 2018-01-12:
@blurfl I can't get any to run without freezing.

madgrizzle, 2018-01-18:
One solution with using makercam is to just use inches and not mm as your units.  When I create the nc file with inches, everything is rounded off to four decimal places.  When I switch the same file to mm, I get 13 or so decimal places.  For example, the first line of the same cut:

Inches:
G0 X-1.5645 Y-8.1566

mm:
G0 X-39.73934010152284 Y-207.17868020304567

GeroBH, 2018-01-22:
Perhaps a sanity-check placed after the -> **File Open** that diverts to a function copied from the py-script? https://github.com/jhessig/metric-gcode-truncator
(This program modifies metric MakerCam generated GCode so as to be usable with grbl CNC firmware.)

BarbourSmith, 2018-01-22:
That file is GPL and python so we could even just incorporate it into GC, right?

blurfl, 2018-01-22:
Someone with more python background can correct me, but the script looks like it does the same thing as out present truncate code - it throws away any digits after the decimal point beyond the maximum, four digits in the case of the script.
It doesn't look like it special-cases G02 or G03 arcs.

thormj, 2018-02-07:
Hrm... I just read that HAAS only accepts 0.0001 inches or  0.001mm.  Half of me thinks that we should ask the firmware the buffer size on startup, and then on load, make sure that all the lines "fit"
If not, we could
1. Warn the user that the file has lines that are too long (and suggest trying different units if they were using MakerCam), and
a. Truncate arbitrarily during the load (and hopefully that will show on the screen as "WTF?"
-or-
b. Figure out how many decimals work for each number on a line and use that number of decimals to do the truncation (best we can do if we're ignorant of the meaning)
-and-ish-
c. Run the g-code during load and, if the destination coordinates are < 0.001 in/mm from the current coordinate, preface that line with a # to comment it out, maybe adding a ##--step too small to the end of the line (this could still blow up if you made a coordinated arc while drilling -- the drill would say "not to close" but when we fixed the line with a/b it could still explode into a circle) -- but if a complete circle was requested, diameter > 0.001 in/mm, we should /know/ about that too and leave the line in place...

c sounds hard and fraught with peril.  b might keep things from exploding better than a.

davidelang, 2018-02-07:
we already do b (with special exceptions handled for arcs that change from small 
distances to full circles if you truncate wrong)

we have talked about doing c

We know the buffer size, the problem is properly filling it without overfilling 
it. Code assistance here would be extremely helpful

blurfl, 2018-02-07:
We've had issues with b) truncating G2 and G3 values causing wildly strange cuts. Maybe instead of truncating at 4 digits we should round up to the fourth digit.

davidelang, 2018-02-07:
On Wed, 7 Feb 2018, Scott Smith wrote:

> We've had issues with b) truncating G2 and G3 values causing wildly strange cuts. Maybe instead of truncating at 4 digits we should round up to the fourth digit.

truncating or rounding can run into the same problem. If you have values in arcs 
that are unique before the action, they make a tiny (invisible) movement. But 
then have them be the same after the action, they make a full circle (because 
the start and end point are the same).

I believe we have code to deal with this already