MaslowCNC/GroundControl #564: Add frame stiffness measurement instructions to calibration sequence

blurfl, 2018-01-12:
Frame stiffness has been shown to be important to accurately calibrating the machine.
 Perhaps we should add something to the instructions of Step3 that would help the user check this during the step. A tape measurement between any two points that would move along with the motor mounts and would not be impeded by the tightened chain would work.

davidelang, 2018-01-12:
My thinking is that before the next set of kits are sent out, we should change 
the frame design to a more reliable one.

BarbourSmith, 2018-01-12:
Yes! I am planning to organize a week where coming up with a new frame design is the main goal of the week. From looking at what has to happen next week I don't think it is a good candidate, but the week after looks good.

I agree that a top bar type design is the way we want to go, but I also want to keep the focus on making a frame which is simple to build and uses universally available materials. 

I think that the goals should be:

1) Low cost, simple, easy to construct
2) Ridgid
3) Require a little dis-assembly of the temporary frame as possible

blurfl, 2018-01-12:
My 2 cents - adding a 10' 2x4 across between the motor mounts would add very little material, and require no extra tools.

krkeegan, 2018-01-12:
Do we want to start an issue in the Mechanics repo for frame suggestions.  I have my own concerns about torsion and the stock frame, in the past I have been in the minority with this concern.

davidelang, 2018-01-12:
On Fri, 12 Jan 2018, BarbourSmith wrote:

> 3) Require a little dis-assembly of the temporary frame as possible

With a top mount design, I'm thinking we can skip the temporary frame entirely.

For example, with the alternate2 frame I have on onshape, everything is cutting 
lengths off of 2x4 material except for the sled (including standoffs and brick 
holders)

Turning this into a permanent A frame would be pretty trivial

The tricky thing is cutting the sled, and I've got some ideas in mind.

David Lang

davidelang, 2018-01-12:
On Fri, 12 Jan 2018, Scott Smith wrote:

> My 2 cents - adding a 10' 2x4 across between the motor mounts would add very little material, and require no extra tools.

We can simplify things further, example attached

davidelang, 2018-01-12:
On Fri, 12 Jan 2018, Kevin Robert Keegan wrote:

> Do we want to start an issue in the Mechanics repo for frame suggestions.  I have my own concerns about torsion and the stock frame, in the past I have been in the minority with this concern.

should we do this on github, or on the forum?

BarbourSmith, 2018-01-13:
I was planning to do it in the forums to get more input, but I'd rather wait for a couple weeks until we've got the calibration thing really dialed in so that I can focus on frame design 100%