MaslowCNC/GroundControl #568: Simulation text question

blurfl, 2018-01-13:
After running a simulation, several text strings are filled in to indicate the distortion calculated. One string refers to the center of the sheet and two of the strings refer to the top corners but none refers to the bottom corners. Is one of the strings actually meant to refer to the bottom corner?
 
[image]

BarbourSmith, 2018-01-13:
Yes, I think that is an astute observation and one of them almost certainly refers to the bottom of the sheet

blurfl, 2018-01-13:
Priobably the bottom one 🙃?

blurfl, 2018-01-13:
I'll close after I remember to get that PR entered...

BarbourSmith, 2018-01-13:
Looking at the code in simulationCanvas.py 

```
       lengthMM = 800
        xOffset = 1100 - lengthMM/2
        yOffset = 500  - lengthMM/2

       #horizontal measurement 
        pointPlotted1, distortedPoint1 = self.testPointGenerator.plotPoint(-lengthMM/2, yOffset)
        pointPlotted2, distortedPoint2 = self.testPointGenerator.plotPoint(lengthMM/2,  yOffset)
        
        #vertical measurement
        pointPlotted3, distortedPoint3 = self.testPointGenerator.plotPoint(xOffset, lengthMM/2)
        pointPlotted4, distortedPoint4 = self.testPointGenerator.plotPoint(xOffset, -lengthMM/2)
        
        printString = "An 800mm square in the top corner on the sheet is distorted\n" + str(lengthMM - (distortedPoint2[0] - distortedPoint1[0])) + "mm horizontally, and " + str(lengthMM - (distortedPoint3[1] - distortedPoint4[1])) + "mm vertically."
        
        lengthMM = 800
        xOffset = 1100 - lengthMM/2
        yOffset = 500  - lengthMM/2
        
        #horizontal measurement 
        pointPlotted1, distortedPoint1 = self.testPointGenerator.plotPoint(-lengthMM/2, -yOffset)
        pointPlotted2, distortedPoint2 = self.testPointGenerator.plotPoint(lengthMM/2,  -yOffset)
        
        #vertical measurement
        pointPlotted3, distortedPoint3 = self.testPointGenerator.plotPoint(xOffset, lengthMM/2)
        pointPlotted4, distortedPoint4 = self.testPointGenerator.plotPoint(xOffset, -lengthMM/2)
        
        printString = printString + "\n\nAn 800mm square in the top corner on the sheet is distorted\n" + str(lengthMM - (distortedPoint2[0] - distortedPoint1[0])) + "mm horizontally, and " + str(lengthMM - (distortedPoint3[1] - distortedPoint4[1])) + "mm vertically."
        
```

I agree that it is the second one which should be the lower square. 

Good catch picking up on that detail!

closed 2018-01-13 by blurfl