MaslowCNC/GroundControl #577: Pressing the 'stop

BarbourSmith, 2018-01-15:


blurfl, 2018-01-15:
If the appropriate fix is changing the text of the button to 'Emergancy\r\nStop', I can take care of it if you want...

BarbourSmith, 2018-01-15:
I want to have a go at trying to find out why it's giving the wrong measurement, I'd like using the stop button to be an option without restarting the calibration process, but if that doesn't pan out let's go with the emergency stop option

krkeegan, 2018-01-15:
Agree, that is odd. is the B10 measure command reading the target and not the current position?

BarbourSmith, 2018-01-15:
That's a good suggestion, that would explain it

blurfl, 2018-01-15:
May be related - I notic ed yesterday that after pressing 'Stop' during a cut run, If I used 'GoTo' to begin the cut at the step before the interrupteion, the Z-axis wouldn't respond to gcode commands (giggled but didn't move). Didn't spend enough time on it to open an issue on it though...

krkeegan, 2018-01-15:
> the Z-axis wouldn't respond to gcode commands

That is weird, I don't have a good explanation for why this would happen.

blurfl, 2018-01-15:
I'll go back and try to come up with enough for a separate issue when time permits.