MaslowCNC/GroundControl #578: Caibration Step9 - make the pictures buttons

blurfl, 2018-01-16:
How about making the pictures of the Quad. and Triang. setups trigger in the same manner as the buttons do?

BarbourSmith, 2018-01-22:
I like this and it shouldn't be too tough. Do we think it could be confusing if we don't make it clear that you can click the picture? Someone could click it and not expect the screen to change.

blurfl, 2018-01-22:
I don't know about that; I just automatically clicked the picture the first time I saw this step. A radio button style choice selected by the picture or the radio button, and a 'Next' button? Starting to get more complicated 🤪

BarbourSmith, 2018-01-22:
Yeah, if you did it instinctively then it's probably pretty intuitive.

I think we can make it happen by applying the [button behavior ](https://kivy.org/docs/api-kivy.uix.behaviors.button.html) atribut to the images like this: https://stackoverflow.com/questions/24668317/kivy-clickable-image-that-replaced-a-button

BarbourSmith, 2018-01-23:
Or way easier we could do it using the the `ButtonTemplate`

object that I created to give the new UI a nicer look. It is used like this:

```
ButtonTemplate:
            text: "Actions"
            funcToCallOnRelease: root.show_actions
            id: actionsBtn
```

It's just an image and a button together. Ok if I assign this to myself and then get your feedback on if what I make is right?

blurfl, 2018-01-23:
Thats the bunny! 
 For extra credit, include pictures of the three triangular options and set the default rotationRadius based on the picture clicked?

BarbourSmith, 2018-01-23:
Yes!

blurfl, 2018-01-23:
I guess we'd need to be careful about overwriting a previous calibration's value, though...