MaslowCNC/GroundControl #583: I can not run on Linux. Tell me please what is the problem.

renat2985, 2018-01-18:
```
[renat@HPeOS:~/Downloads/GroundControl-master]
% python --version                                                                                                                  2 ↵
Python 2.7.6
[renat@HPeOS:~/Downloads/GroundControl-master]
% pip install -r requirements_linux.txt
Downloading/unpacking appdirs==1.4.3 (from -r requirements_linux.txt (line 1))
/usr/share/python-wheels/requests-2.8.1-py2.py3-none-any.whl/requests/packages/urllib3/util/ssl_.py:100: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  Downloading appdirs-1.4.3-py2.py3-none-any.whl
Downloading/unpacking Cython==0.25.2 (from -r requirements_linux.txt (line 2))
  Downloading Cython-0.25.2.tar.gz (1.7MB): 1.7MB downloaded
  Running setup.py (path:/tmp/pip_build_renat/Cython/setup.py) egg_info for package Cython
    Unable to find pgen, not compiling formal grammar.
    
    warning: no files found matching '*.pyx' under directory 'Cython/Debugger/Tests'
    warning: no files found matching '*.pxd' under directory 'Cython/Debugger/Tests'
    warning: no files found matching '*.h' under directory 'Cython/Debugger/Tests'
    warning: no files found matching '*.pxd' under directory 'Cython/Utility'
Downloading/unpacking Kivy==1.9.1 (from -r requirements_linux.txt (line 3))
  Downloading kivy-1.9.1.tar.gz (16.4MB): 16.4MB downloaded
  Running setup.py (path:/tmp/pip_build_renat/Kivy/setup.py) egg_info for package Kivy
    Using distutils
    
    Cython is missing, its required for compiling kivy !
    
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_renat/Kivy/setup.py", line 184, in <module>
        from Cython.Distutils import build_ext
    ImportError: No module named Cython.Distutils
    Complete output from command python setup.py egg_info:
    Using distutils

Cython is missing, its required for compiling kivy !

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_renat/Kivy/setup.py", line 184, in <module>

    from Cython.Distutils import build_ext

ImportError: No module named Cython.Distutils

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_renat/Kivy
Storing debug log for failure in /home/renat/.pip/pip.log
[renat@HPeOS:~/Downloads/GroundControl-master]
% python main.py                                                                                                                    1 ↵
Traceback (most recent call last):
  File "main.py", line 6, in <module>
    from kivy.config                import Config
ImportError: No module named kivy.config
[renat@HPeOS:~/Downloads/GroundControl-master]
%             
```

blurfl, 2018-01-18:
I see "Cython is missing, its required for compiling kivy !" in the error output. Have you been through the steps in the Linux section [here](https://github.com/MaslowCNC/GroundControl/wiki#gc-installation-guides)? Did those steps go smoothly, or were there problems getting those required files installed?

GeroBH, 2018-01-18:
What Linux are you running? On Ubuntu Mate, Lubuntu, Lubuntu and Man Jaron (arch) I never used the requirements.txt. Did not work for me in VM way back, so I just followed the instructions here https://github.com/MaslowCNC/GroundControl/wiki/Linux

closed 2018-01-19 by renat2985