MaslowCNC/GroundControl #584: Size 'Calibration Test' pattern to the workarea dimensions

blurfl, 2018-01-18:
PR #579 adds a Calibration Test file, but that pattern is only useful on a standard sized workarea. How about calculating the pattern to suit the workarea? This could be separate choice if there is a reason to have the standard pattern available as well.

BarbourSmith, 2018-01-18:
I do like the idea of having a standard test pattern so that we're all comparing apples to apples, but I think a second version which sizes for the work area would be a great idea

reecej, 2018-01-19:
This could be implemented similar to the new triangular kinematics calibration cut pattern. The locations of the cuts are based on the work area dimensions, so that it scales with different sized machines.