MaslowCNC/GroundControl #586: Calibration Step10 Triangular mode disables text entry on error input

blurfl, 2018-01-20:
I fat-fingered a value and clicked 'Enter Values' and the Measurement input boxes became disabled (though the 'Bit Diameter' box was still available. Te only way forward was to 're-cut the test pattern'. What would it take to make it possible to re-enter the values witout re-running the cut?

reecej, 2018-01-20:
@blurfl did it accept the value? If the value entered was completely wrong, the basic error-checking would kick it back to you. If the error-checking didn't kick it back, then as the code is written today you would need to re-cut the test pattern, with the option of pressing the stop cut button right after you press re-cut test pattern.

Would it be better if you were able to re-enter measurements if calibration wasn't successful? This would make sense, as since the machine parameters are not changed (due to lack of successful calibration), there should be no need to re-cut.

davidelang, 2018-01-20:
On Sat, 20 Jan 2018, reecej wrote:

> Would it be better if you were able to re-enter measurements if calibration 
> wasn't successful? This would make sense, as since the machine parameters are 
> not changed (due to lack of successful calibration), there should be no need 
> to re-cut.

The only problem I see with re-entering the numbers is that the initial rotation 
radius is part of the calculations

I'm wondering if it makes sense to always set the rotation radius to a fixed 
value when doing the calibration (both the cuts and the calcuations).

I wonder if we could set it to 0 or if that would result in cuts going too low 
(especially with something like the 7.5" rotation radius of the large-router 
version of my kit)

reecej, 2018-01-20:
Well, the change I made in that PR was to allow the values to be re-entered without a re-cut if a successful calibration was not performed. In that case, no parameters have been updated, and if the re-cut was performed it would be identical to the previous cut pattern, and therefore not worthwhile.

If the calibration was successful and the parameters were updated, then this changes a bit. Like David mentioned, we can either use the current values to help maintain the sled on the workspace, or use default parameters which would not necessitate a re-cut.

My preference would be to use the closest values to the actual measurements as possible, as that allows us to push the sled to the borders of the work area more reliably, and therefore experience more chain sag and make more accurate guesses for the machine parameters.

blurfl, 2018-01-20:
> did it accept the value? 

The error detection worked properly, the value was not accepted, no calculation was completed. I measured from the 40cm mark on my tape and forgot to subtract, so it was proper to trap it. If a user enters a value with a much smaller error, you couldn't expect to recognize that. Let's not get to the "Are you sure? Enter y/n" level of user input here 🙂...

blurfl, 2018-01-21:
It does indeed! I'll get it tested tonight or tomorrow morning, sorry to take so long when you jumped up and write a fix practically instantly 👍🏻💯

davidelang, 2018-01-21:
how do you define 'expected'?

depending on the version of the linkage kit that someone is using, the rotation 
radius can vary from 5 to 9 inches (more if they are using an unusual kit to 
deal with a large router)

closed 2018-01-22 by BarbourSmith