MaslowCNC/GroundControl #591: Load/Align a "background image" on the canvas

thormj, 2018-01-25:
Enhancement: Be able to load a "background image" onto the g-code canvas and align it so you can see what the cut looks like on a picture of stock in the machine.

I've gotten a process to work (using a T3i camera, some reflectors, and OpenCV
But I need some guidance on how the menu/buttons work (thinking about adding a button to the right of the Settings button called "Background"... we can move/integrate it better later, but...)

So can someone point me to the direction of how the "Actions" button calls up the "Actions" popup that has the other menus tacked together?

Raw image:
[image]

Aligned image:
[image]

On GroundControl:
[image]

To do this, I need to know:
  Positions of 4 markers on the machine (TL,TR,BL, BR) - ok if outside workspace (I put mine 5cm outside the 4x8 area)
  Colors of said markers in OpenCV's HSV space (I'll make a picker)
  Contrast adjustment that should be applied to make it so you can see your cut lines
  Where to load the image from?

thormj, 2018-01-25:
I'll have more notes as I go along, but duplicating a color was a mistake (not mathematically, but half the time the corners come in 1234 and the other half 1243, so... more fixes).

reecej, 2018-01-25:
This is great! I really like the idea and execution.

Buttons are defined in the Kivy code: [groundcontrol.kv](https://github.com/MaslowCNC/GroundControl/blob/master/groundcontrol.kv)

Look around line 274 and you'll start to see it.

thormj, 2018-01-25:
I see the menu being declared there, but the part I'm not getting is how it goes from
  groundcontrol.kv:40 [root.show_actions] to
  [Whatever the container is for the buttons on that screen] 

I see all the menus being declared (and I even added a button to that one), but I don't see [what it is that's holding those menus together] or where root_show_actions() is located (then again, I've been having an argument with grep today so maybe it'll be better in the morning).

Kivy really doesn't like a button that doesn't have a callback assigned to it (I'm assuming that's why it's really uglified in my screenshot)... but I don't know where to put my function/menu/widget to satisfy it,  But I'm stoked.... this is really neat!

reecej, 2018-01-25:
[screenControls.py](https://github.com/MaslowCNC/GroundControl/blob/master/UIElements/screenControls.py) might be the missing link. show_actions is defined there, and there it sets the OtherFeatures Kivy menu to be loaded. Most buttons reference a Python function, which has the ability to interact with the Kivy menu.

If you wanted to make a button to say load this image, you likely would want to make the button in the Kivy menu and have it call a Python function when pressed.

thormj, 2018-01-25:
Aha!  Thank you!  Onwards!

thormj, 2018-01-25:
Ok... I'm not sure how to ask for help with GitHub, but I thought I put everything in place (well, in UIElements/Background.py, it should explode because I don't have backgroundMenu yet) to link a new button in, but it doesn't start.  Python says "ScreenControls has no attribute 'open_background'... but I thought I defined that on UIElements/ScreenControls.py:46

Help!  I've written and it won't come up!
This is the https://github.com/thormj/GroundControl ImageCanvas branch

reecej, 2018-01-25:
It looks like your indentation is wrong. In Python your indentation is context-relevant, so right now your function is being defined under close_popup because it's indented under there. Modify your screenControls.py so your open_background function is indented similar to the other functions.

thormj, 2018-01-25:
Thank you!  That kept escaping me... not I've changed my editor to change tabs into spaces so hopefully I don't do that again.

BarbourSmith, 2018-01-25:
This is a very cool idea! I am excited to incorporate it 💯 👍

BarbourSmith, 2018-01-25:
Some of the features shown here seem relevant:

https://www.youtube.com/watch?v=vgPxbyLc7k4

This video might cover how things all link together, but it's a bit dry:

https://www.youtube.com/watch?v=KkpdSb68yuo

thormj, 2018-01-26:
I've posted all the bits to my branch (everything is hardcoded and in the wrong place right now, but I figure some people might want to see what I did): https://github.com/thormj/GroundControl/tree/CanvasImage 

Don't pull yet... beyond the several ToDo's in the code, everything is hard-coded to my test computer.

If someone could help me with Kivy (again... sheesh.  I seem to be on Repeat), I have the menu button working... but no buttons appear on the popup menu (you can close it with ESC).  I thought I had everything for the buttons to appear (yet crash when pressed because the code in UIElements.BackgroundMenu.py are not even close to being done), but nope.  I have a pop-up canvas, and the buttons are in the .kv, but there seems to be a missing linkage (as usual, I have a screw loose somewhere)...

I'm thinking of putting the "find corners and warpimage" into the re-load method and storing the "texture" into the datastructure (like the g-code),   That way, refreshes can be fast (err... fast...er)...

reecej, 2018-01-26:
Nice process so far! I haven't taken it for a drive yet, but I'm liking the code and progress so far!

Regarding your menu problem, it looks like your Kivy menu is defined as "BackgroundMenu" whereas your Python is calling "Background". Switch the Python content reference in screenControls.py to "BackgroundMenu" and I think you should be good.

thormj, 2018-01-26:
I've got a file called Background.py, but I couldn't get lines 12/13 to work.
I was patterning it after the View / ViewMenu but I missed something...

I think I've got the guts running w/o todos, so now it's time to work on the integration...

reecej, 2018-01-26:
The screen that isn't showing up, is it the screen with the Open/Update/Remove Background buttons?

thormj, 2018-01-26:
Yep that one.  I tried to define it .kv...

reecej, 2018-01-26:
Yeah, so in screenControls.py your content reference isn't to your Python file, it's to the code block definition in Kivy, which is BackgroundMenu. So your screenControl.py line 50 need to change to BackgroundMenu. Then Kivy calls your Background.py.

thormj, 2018-01-26:
Ok, if I do that, it comes up, but when I hit the button, it crashes out with "global name 'BackgroundMenu' is not defined.

reecej, 2018-01-26:
It's looking like you're missing the module imports for it.

Step 1: Change content reference in screenControls.py to BackgroundMenu.

Step 2: Add the following line to the import sections of main.py and screenControls.py:
import UIElements.backgroundMenu import BackgroundMenu

And give it a try. 👍

thormj, 2018-01-26:
Oh man, thank you!  That is beautiful... now to get it all wired!

thormj, 2018-01-27:
Ok, I have it wired and in place and working properly (as long as you put the right data in the data.py file)... but I need some help on the BackgroundSettings menu (ok, I'd really like somebody who knows Kivy to do it and for me to admire it and hook code into it).
I need a dialog that looks like this:
[image]

thormj, 2018-01-27:
I added lightening/darkening buttons (https://github.com/thormj/GroundControl/tree/CanvasImage aeb145f..ee5d1b7), but 2 bugs:
  1. The buttons are wide, despite what I put in groundcontrol.kv - why?
  2. I can't set the background/text on screencontrols.py:43 -- I thought I just copied/pastied...

So I need more kivy help in addition to the "Dialog Box Request"...

reecej, 2018-01-27:
I can take a look this weekend. I've only been using Kivy for about a month, but am happy to lend any knowledge I have! I am having a little difficulty understanding your image though. The top has four text boxes next to each button? And what is on the left side of the screen?

thormj, 2018-01-27:
The top bar represents the 4 corners, and each corner has 4 "bits" - X,Y, HSVmin,, HSVmax (and a radio button to pick which corner the stuff on the left is modifying).
On the Left side of the screen, there's 2 boxes ("Position") for X/Y [though that label might be misleading -- the position is the coordinates from center [and I suppose it should obey inches/mm]), and 2 ColorPickers for HSVMin/HSVMax, and 4 buttons (Test, Load, OK, Cancel) and a Label that shows the result (1/2/3/4 dots).

It's clunky, but that's what I need to set for each of the corners...
The usage pattern is
1. Go here, load a pic
2. For each corner, 
2a. Put the coordinates in manually
2b. Get the HSVmin/HSVmax right for that corner, using test button so you get a good circle on it
3. When good, the crosshairs down the middle should line up with the center.  if not, tweak the xy postion for each corner until it does.
4. Load another pic and repeat 2b, tweaking HSVmin/HSVmax so that all corners in all pics are good
5. Hit Ok, and you're loaded
6. Load a background from the prior menu
7. Observe the center (you see in my pic, I put a dot on the center of the backboard); if it doesn't line up with the crosshairs when Maslow is at (0,0), then tweak 2a until it does (but it should be if you did step 3.

Hopefully by then I'll have the data saved in the .ini file...

reecej, 2018-01-27:
Sure thing, I think I can help with that!

A little brainstorming. I like the idea of the reflective markers, but am worried they might be more difficult for the image processing to pick up in a more cluttered shop area (like mine). It also requires several extra steps, which I'm assuming once set up won't be required but might be difficult the first time.

Have you considered a version where rather than the reflective markers, the user just clicks the four corners of the image with their mouse, which are then accepted as the corner positions? I believe I could write that Kivy code for you a well, which then you could feed into your awesome image manipulation code to skew the image properly, etc.

The work flow I imagine for this would be:

1. Load pic
2. Be prompted to click each corner in the image
3. Click top left
4. Click top right
5. Click bottom left
6. Click bottom right
7. Verify image aligns with work area, if not re-select corners
8. Hit ok, other steps are the same

It seems like that would present a simpler user experience, but I'm not sure if it would give you the info you need.

thormj, 2018-01-27:
That will work too.  I'll figure out how to split the code to accomodate it.  I'm actually going to try LED lights this weekend... the HSV values change a lot depending on flash/no-flash with the reflectors -- the hope is that you'll be able to point a webcam or rig a DropBox filesharing thing to a directory and have it "just work" (new pic, new background), but the manual align with "click-the-dots" would be great for bad lighting (I even know just the place to put it -- there's a spot "what if we don't find 4 dots").

reecej, 2018-01-27:
I like the idea of LED lights, and personally would go that route if given the option. But I like the idea of having a backup method where the user clicks the corners. That way, if a user wants to use this feature, they can install the LED lights or select the positions manually.

servant74, 2018-01-27:
Sorry to but in, but I am trying to follow this.  What is the major goal of
this?  To put an alligned image of a cut area on the Ground Control
display?  I am curious, what does it buy the user? ... Thanks for your
patience, I am just trying to understand.

reecej, 2018-01-27:
@thormj's idea is to allow the user to align new pieces to be cut on a partially-used piece of plywood, for instance. So the user could take an image and have it aligned to the work area in GroundControl, then use this information to visually locate the new pieces to avoid the previously cut out areas.

thormj, 2018-01-28:
Yep... here's a better pic:
[image]
If you move the sled over to the spare area on the top-left... set home and see if it fits.

thormj, 2018-01-29:
Ok... Logic is in place... bring on the critique (especially on my problematic python skills- I really am not sure where I should've put all the pieces [in self, self.data, or ?... it seems that if it's not in self, data can't be shared between UI callback functions, but when the dialog evaporates, there's a lot of objects that aren't useful, so if they were garbage collected...])  :)
github.thormj/groundcontrol 22a9bea..7eab38f (or later)

I truly stink at UI's... I tried to make a simple BackgroundPointPickDlg thing (a label, 2 buttons, and a canvas), but nothing is showing up where it should... so... even more help requested (I don't know how github works, so I guess I'm asking for help on my branch/repo, then I'll pull it all into mine, and then I'll make a pull request to pull it from mine to here?).

I did figure out a way around my +/- button sizing and color: I cheated... I set every button to the same background and text color (see the SecreenControls.py file).

Now off to figure out how to put my globals into the ini file...

thormj, 2018-01-29:
Ok, now I'm confused - can someone help me figure out the user settings?
It looks like (based on the openFile/gcodeFile), I need:
1. Variables in the Datastructures\data.py
2. Mimics of those variables in the Globals datastructure global_datastructure.py 
3. a block in the json structure in main.py (referencing the globals)
4. a block in the "Load User Settings" of main.py (around 469) to copy data from self.config.get to self.data
5. a block in the build.config of main.py (around 500) for setdefaults()
6... but I don't see where self.data.gcodeFile gets written to save buffer anywhere (but it works, so...)

So
1. Do I have that right?
2. How does the system know to write "the latest" gcodeFile to openFile?

thormj, 2018-01-31:
The good news is that I have all the bits in place to load the settings :).
The bad news is that Python and I are not getting along:

I added code to the RunPeriodically in main.py to check for new backgroundFiles, but I can't seem to import UIElements.BackgroundMenu into main.py -- it's giving me an ImportError.

Help: https://github.com/thormj/GroundControl/tree/CanvasImage d64e097..3853f1a

thormj, 2018-02-06:
Python and I are now almost getting along.  I didn't put the RunPeriodically code back in (so I still need help there), but I did get the Manual PointPicker to work.  This should be (manually) usable by everyone as long as the X/Y of the alignment points match my machine.  To really fix, we need to get the settings dialog up-n-running (and I need help there too).

thormj, 2018-02-06:
Need help:
1. Data is not saving (you always go to a weird place when loading an image after a restart).
2. For some reason, the json.dumps is not initializing the properties in BackgroundSettingsDlg, so it crashes when it comes back.

 01a60cb

thormj, 2018-02-07:
PBCAK issues fixed with the settingsdialog (to self or not to self, that is the question)
Feature complete and saves the data... so I'll make a pull request