MaslowCNC/GroundControl #592: Display velocity for Maslow and current g-code line on the screen

thormj, 2018-01-27:
Figuring between Z and not connected, I'd like an indicator that says:
Vel: 30 ipm  | 30 ipm

Where the first number is from Maslow, and the 2nd number is from the current line of g-code.

thormj, 2018-01-30:
Is there a (preferably auto-generated) list of g-code commands somewhere?

thormj, 2018-01-30:
I'll take this one and run with it for a while...

BarbourSmith, 2018-01-30:
I believe that there is a list of supported commands on the wiki page, but it is not automatically generated (which would be slick)

We've been talking about setting up circle CI integration to automate tasks like that which I am happy to help with on my end

BarbourSmith, 2018-02-01:
I think you nailed this one, good to close?

thormj, 2018-02-01:
Yep.

closed 2018-02-01 by thormj