MaslowCNC/GroundControl #594: Calibration: Measure Top of Board - Start from 0 instead of retracting

thormj, 2018-01-28:
Add a button to "Start from zero" (put tooth at 12:00 and first link on top) and extend chain.
Faster than retracting it 2.6m...

blurfl, 2018-01-28:
This step gives a very crude measurement - how about asking the user to use a tape measure (with suitable pictures) instead?

davidelang, 2018-01-29:
this step is eliminated with triangular kinematics.

thormj, 2018-01-29:
Err... if you don't do that, how do you figure out what the Y of center is?

davidelang, 2018-01-29:
very interesting math that figures out how far above the work area the motors 
are by comparing different triangles

blurfl, 2018-01-29:
The triangular calibration step calculates the distance of the center cuts from the line between the motors. When the user enters the distance from the top cut to the top of the workarea, that value is subtracted from the calculated distance of the top cut to give the distance from the motors to the top of the workarea. Calculating the distance from the motors to the center of the workarea is left as 'an exercise for the student' 😁.

BarbourSmith, 2018-01-29:
I vote for the "ask someone to measure it" option. We just need a rough measurement and retracting all that way is a hassle

thormj, 2018-01-29:
That works... it just seemed to take a loong time...

bdillahu, 2018-02-06:
I had just logged in to recommend swapping the "measure top of workspace" with the "measure distance between motors". You're starting at zero to do the distance between motors... pay out enough and measure to top of board, then keep paying out to measure between motors. Then you don't have the "pay all the way out, nearly all the way back in, then way back out again".

Or just use a tape measure :-)

BarbourSmith, 2018-02-06:
I totally mean measure it with a tape measure 😀 

I guess I should have been more clear because of course all the options involve measuring the distance 🙄