MaslowCNC/GroundControl #596: On the "Extend Chains" cal step , have a move-to-zero button and +/- Y to adjust 0

thormj, 2018-01-28:
During my last cal, on the "Extend chains step", I noticed that when I first hooked them up, the sled was considerably lower than the center I was expecting... only to come up to closer zero (probably due to stretch) when I left the screen.

I'd like a button for "move to zero"  on the screen... and preferably a couple of buttons to adjust Y0 up/down.

BarbourSmith, 2018-01-29:
I think this is automatic, the sled should move to center as soon as you leave the step. It won't be the exact center because the calibration process hasn't been run.

thormj, 2018-01-29:
It does (and afaict, this zero doesn't move during the cal step, but I'll check [quad user here]).  But when I first hooked the chains up, I was dismayed that it was about 2.5" below the center.  When I went to the next step, it went up to only 0.25 below the center (which is why I wanted the "Adjust Y" here... so I can put center where I want it with the router in place (so the crosshairs 0,0 aligns with the master-mark on my board and then I can align my image processing so that the pic of the master-mark aligns with the crosshairs @0,0).

BarbourSmith, 2018-01-29:
Oh, got it!

That .25 inch error will be further refined in the last step of the calibration process so any adjustment made here would be overwritten later.

What about a "define center of sheet" button somewhere outside the calibration process?

thormj, 2018-01-30:
That would be excellent.  Suggestion - put it on the "Extend Chains" dialog (since you're extending them after a poweroff anyway" (is that the same screen?).