MaslowCNC/GroundControl #597: Add an Abort to calibration sequence

thormj, 2018-01-28:
There's no obvious/easy way to abort calibration (eg, I did 90%, but then realized I forgot to hook up my Z axis... and just plugging it in didn't work).  So I didn't want to overwrite the values...  Not sure where it saves the info to the machine, but there's no way to get out without going to the end and saying "close."

blurfl, 2018-01-28:
I agree you should be able to back out (or skip forward, I think) without anything getting changed unless you use the 'Continue' button (rename it 'Accept Values'?). Presently, if any values have changed in earlier steps, GroundControl.ini gets re-written when the step is entered. Bug or Enhancement? 

Currently the z-axis step comes after the 'review and accept' step. I think that was by design 🙂. Changes to the 'z-axis present' switch in that step seem to work as expected. I unhooked my z-axis and went to that step and clicked a movement button to try to replicate the error. After I plugged the motor back in I was able to use the movement buttons as expected. If you have a way to replicate your error situation, that sounds like a good candidate for a new issue.

thormj, 2018-01-29:
I'll double check if it's truly an error and will create a new issue for "plugging z in halfway through calibration"...

I'd still like a "Panic-Quit-Abort" button for if you had done a calibration previously, but you b0rked something this time around (like trying to extend the chain 3.8m ...)

BarbourSmith, 2018-02-01:
Should be good on this one!

closed 2018-02-01 by thormj