MaslowCNC/GroundControl #598: Add "Estimated Time Remaining" to G-Code section

thormj, 2018-01-29:
Enhancement:
It's probably complicated and might be inaccurate (just ask the 3D Printer folks), but I'd like some kind of Estimated Time Remaining readout...

Is there a way I can pre-mark issues as enhancements?

blurfl, 2018-01-29:
Is the 'Labels' drop-down accessible to your account? Maybe not, I'll get it!

davidelang, 2018-01-30:
On Mon, 29 Jan 2018, Thor Johnson wrote:

> Enhancement:
> It's probably complicated and might be inaccurate (just ask the 3D Printer folks), but I'd like some kind of Estimated Time Remaining readout...

it's been asked, very complicated and inaccurate (much more so than the 3D 
printer estimates due to the long distances and slow movement)

> Is there a way I can pre-mark issues as enhancements?

see tags