MaslowCNC/GroundControl #603: Machine won't start again after pressing stop

BarbourSmith, 2018-02-01:
For some reason my machine has stopped starting again after the stop button is pressed. It will stop immediately, but not move again until GC is restarted. 

This is a new behavior.

BarbourSmith, 2018-02-01:
I think this is actually a firmware issue. I'm not seeing an OK command returned after the stop command is sent to indicate that it has been processed

BarbourSmith, 2018-02-01:
Or maybe it's that we're getting one 'OK' instead of two.

From the log:

`Sent: !
ok
`

But when we print the amount of space GC thinks is available in the buffer it looks like this:

[image]

So there's `236 `spaces in the buffer because the line of gcode hasn't been processed yet, then there are `234` spaces because we've sent the "`!\n`" command, then we go back to `236` when it is marked that the stop command has been processed...so do we need to send an extra "OK" after stops are processed to indicate that the extra line has been cleared from the buffer? It has been cleared.

BarbourSmith, 2018-02-01:
I am going to close this and re-open it in firmware

BarbourSmith, 2018-02-01:
As expertly summarized by @krkeegan :

> oooh ooh, I think I understand what you are describing now.
> 
> So let me see if I have this right.  
> 
> 1. Send some command "G0 X5"
> 2. While the command is executing send "!"
> 3. Firmware sends "ok" response to "!" command
> 3b. GC sees the "ok" recognizes that it was in response to a quick command and clears only that portion of the buffer.
> 4. Firmware bails out of routines that it is running
> 5. Firmware dumps contents of serial buffer
> 6. Firmware waits for new commands
> 7. Meanwhile GC is still waiting for confirmation of "G0 X5" command
> 8. aaaaand we are stuck
> 
> I think I get it.
> 
> I see two solutions:
> 1. The firmware sends the necessary "ok"s to clear everything
> 2. GC recognizes that a stop dumps the buffer contents on the machine and clears it's own tracking buffer when it receives the stop "ok"
> 
> I think this is all a matter of perspective and what we want an "ok" message to mean.
> 
> I am inclined to say GC should handle this, that the firmware is correct in not sending an ok to the "G0 X5" command because that command never successfully completed.  Plus this design would allow us to queue more than one line on the machine (which I think we can do eventually) otherwise if we had multiple lines, we would need the firmware to say "ok" to those extra lines in the buffer, even though it hadn't even started performing them.
> 
> Thoughts? I am open to considering the alternative.

thormj, 2018-02-07:
Is there a standard in g-code for what the ? command returns?
It might be a break with tradition, but I like having the machine spit out "I'm running"/"I'm waiting"
Maybe <x,y,z,> for "moving" and [x,y,z] for "resting at"?
(and if you used ~ to pause motion, I would assume that it would still be "moving" as the command had been given, but if you  ! to kill the motion and the command, it would soon go to "resting".. maybe we need another flag for "paused at"?)

thormj, 2018-02-09:
FWIW, it looks like grbl can return all manner of things, but following its report.c was being incompatible with my brain for a bit.  You can ask it for status and it sends Idle/Run/Hold/Alarm -- it looks like it says that in front of the coordinates like
<IdleAlarmDoor:3|?pos: axispos |Bf: blockbuf,rxbuf |Ln:line ...>
So it looks like we're getting ahead of ourselves when we report our position before the |?pos: text

BarbourSmith, 2018-02-13:
Is there any reason this is still open? I thought we got it in #606

blurfl, 2018-02-13:
Sounds like the issue recently brought up in the forum was on a version before this merge. I tried the steps described in that issue on tis version and found no failure. I'd say 'close it' 👍

closed 2018-02-13 by BarbourSmith