MaslowCNC/GroundControl #614: Ground Control runs, but renders incorrectly

BarbourSmith, 2018-02-07:
My four year old netbook with a 32gb hard drive finally go so full that windows 10 couldn't update anymore because there wasn't enough space on the hard drive to download the update, so I caved and bought a new low end computer (an HP 14-bs152od). After running through the setup instructions GC runs well, but it renders strangely.

[image]

Has anyone else seen an issue like this?

While I am working through updating drivers and such, @blurfl  and @krkeegan will you guys go ahead and start merging the PRs that are piling up? I might not be able to test anything today and I think we should keep things moving.

BarbourSmith, 2018-02-07:
I see the same behavior when running the release version so I don't think the issue is my Kivy install

blurfl, 2018-02-07:
Wish we had that Linux thumb-drive version 😁 of theGC environment...

blurfl, 2018-02-07:
Are any of the PRs tested enough for merging? In what order would you like the ones needing testing addressed?

BarbourSmith, 2018-02-07:
It looks to me like #608, #609, and #610 are good to go. #611 is more of a UI experience question so not having seen it I can't comment, but I support the idea behind it

krkeegan, 2018-02-07:
don't do #608 yet, is corresponding firmware change has some glitch with the stop button, I am working on it.

blurfl, 2018-02-07:
Sounds like 609, 610 and 611 then. @krkeegan, do you want to pull the trigger on these?

krkeegan, 2018-02-07:
haha, why me? but sure I can.

blurfl, 2018-02-07:
Well done, sir, and speedily too! 👍🏻💯

krkeegan, 2018-02-07:
Alright, everything I am comfortable merging is done, and seems to work.

BarbourSmith, 2018-02-13:
I ended up returning the computer which had this issue because I finally got my old one working again (nothing wrong with it, just a small disk) so I can't test this anymore and I'm going to close the issue

closed 2018-02-13 by BarbourSmith