MaslowCNC/GroundControl #617: z-axis popup can exit without using the 'Done' button

blurfl, 2018-02-08:
- open the z-axis popup, change the units. Note that the main screen units change as well. Close the popup using the 'Done' button, the main screen changes back to the units it had before the z-axis popup opened (thanks to PR #611 👍💯). If instead you leave the z-axis popup bu clicking outside the window, the units on the main screen do not change back. This leaves the machine interpreting the gcodes in the wrong units.

thormj, 2018-02-14:
The machine is in funky state has been fixed, but I just found out how to make kivy dialogs modal (so they won't go away on their own) - ` auto_dismiss=False` in the Popup constructor.

Should we make the zPopup modal?

blurfl, 2018-02-14:
I'd say it's not necessary, now that the z-axis popup doesn't mess with the main units. Let's leave it be for now?