MaslowCNC/GroundControl #623: Kivy: A click "goes-through" to the new menu/dialog

thormj, 2018-02-13:
I think a Kivy bug, but I haven't seen any good way to handle it.
If you click a button, the widget that pops up will get the `on_touch_up` event when it pops up under the cursor.  Eg, if you're using the file dialog, it seems it will always get "an extra click" (the click came from the "Load File..." button).

We could deal with it by changing everything to `on_touch_down` (as that already occurred and won't occur again), but then you can't click on an item... say no, nevermind, and drag away to _not do it_.  And every example in Kivy seems to be geared to the `on_touch_up event`.

BarbourSmith, 2018-02-13:
Hmmm...what OS are you seeing this on? I'm don't think I've run into it before

thormj, 2018-02-13:
I was dumb.  You should tie events to on_touch_up (as the examples say), otherwise touch_down->file dialog->touch selects file.,, so I added it to the notes,

closed 2018-02-14 by thormj