MaslowCNC/GroundControl #624: Clicking the Inch/mm during a cut teleports the sled

GeroBH, 2018-02-13:
I love the way the Z settings is locked (disappears).
Can this be done for the Inch/mm button as well during a cut?

This observation is from a test system without shield, so might not be applicable on a real system.
Tested with Fake Server activated and without.
To replicate click the button during a cut. The first 2 times as expected there is a jump of the position indicator according to the difference, like 10mm to 10 inch. The third or more click will catapult the indicator to outer space, so you need to zoom far out to find it.
My concern is that the button is right above 'pause' and 'stop' and not only on touch based systems can be hit by accident.

davidelang, 2018-02-13:
the inch/mm button (and arrow movement buttons) should not be active while 
running a g-code program. When the program is paused (or blocked for Z axis 
movement), the movement buttons should work, but the in/mm button should only be 
available when you are not in the middle of executing g-code

BarbourSmith, 2018-02-13:
Great point guys. I believe that locking the z-axis like that was an excellent and recent addition (great work!). I don't think it should be hard to do the same thing to the other buttons at all.