MaslowCNC/GroundControl #625: Division by zero in speed calculation

BarbourSmith, 2018-02-13:
I'm seeing GC crash occasionally with this error. I'm not sure why we're getting a time of zero, but my bet is it's safe to catch it with a try...except which prints a warning.

```
[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 426, in <module>
     GroundControlApp().run()
   File "C:\Python27\lib\site-packages\kivy\app.py", line 828, in run
     runTouchApp()
   File "C:\Python27\lib\site-packages\kivy\base.py", line 504, in runTouchApp
     EventLoop.window.mainloop()
   File "C:\Python27\lib\site-packages\kivy\core\window\window_sdl2.py", line 663, in mainloop
     self._mainloop()
   File "C:\Python27\lib\site-packages\kivy\core\window\window_sdl2.py", line 405, in _mainloop
     EventLoop.idle()
   File "C:\Python27\lib\site-packages\kivy\base.py", line 339, in idle
     Clock.tick()
   File "C:\Python27\lib\site-packages\kivy\clock.py", line 581, in tick
     self._process_events()
   File "kivy\_clock.pyx", line 367, in kivy._clock.CyClockBase._process_events (kivy\_clock.c:7700)
   File "kivy\_clock.pyx", line 397, in kivy._clock.CyClockBase._process_events (kivy\_clock.c:7577)
   File "kivy\_clock.pyx", line 395, in kivy._clock.CyClockBase._process_events (kivy\_clock.c:7498)
   File "kivy\_clock.pyx", line 167, in kivy._clock.ClockEvent.tick (kivy\_clock.c:3490)
   File "main.py", line 284, in runPeriodically
     self.setPosOnScreen(message)
   File "main.py", line 393, in setPosOnScreen
     self.frontpage.setPosReadout(self.xval, self.yval, self.zval)
   File "C:\Users\Bar\Git\GroundControl\UIElements\frontPage.py", line 128, in setPosReadout
     Vel = delta / (time()-self.lasttime) * 60.0 #In XXXX/minute
 ZeroDivisionError: float division by zero
```

blurfl, 2018-02-13:
How to replicate it?

BarbourSmith, 2018-02-13:
I've seen it twice now, but I seems to be random 🙁

closed 2018-02-13 by BarbourSmith