MaslowCNC/GroundControl #628: alter GroundControl textConsole to preserve and show more lines

blurfl, 2018-02-15:
I like to be able to scroll back in the textConsole portion of the frontPage window to look at earlier lines. I've done this locally by stretching the textConsole portion of the window so that it uses the full distance from its current position to the bottom of the window, and changing the way it is updated to preserve 2000 characters? instead of just 500. Is anyone else interested in this? 

I could also wish that scrolling the textConsole did not affect the main image, but I haven't looked for a way around that. Any hints about how to de-couple the two?

BarbourSmith, 2018-02-15:
I support this. I think I reduced the number of lines saved because it was getting slow, but we've made a lot of improvements to the UI since then so I say we go for it in upping the number of lines.

As far as the scroll issue I know exactly what you are talking about. It's an issue when scrolling in the settings menu too or when looking at a gcode file. Basically the scroll wheel always zooms even if you can't see the canvas which isn't the right behavior

blurfl, 2018-02-15:
I've tried the bigger textConsole on various screen sizes in OSX and Win10, but don't currently have a way to try it on very small screens (I see that main.py sets the minimum screen size to 620x440! Is that an Android phone, or a RasPi touch screen?)

BarbourSmith, 2018-02-15:
I bet that GC won't look great on a 620x440 display 😀 

I think that resizing the window on any platform will simulate the effect of a smaller screen, but I am no 100% sure about that

closed 2018-02-17 by blurfl