MaslowCNC/GroundControl #629: Set z-axis 'safe travel' in Settings page

blurfl, 2018-02-15:
Currently, the ‘Home’ button, ‘Return to Center’ button and Z-Axis popup 'Save and Raise to Traverse' button in GC have hard-coded values for safe travel. 
Add a value to maslowSettings.py in the 'Maslow Settings' section to let the user choose the value and save it between launches.
Question: other settings in this section are in mm and require the user to enter the value in mm. Should this value be the same? I lean toward 'yes'...

BarbourSmith, 2018-02-15:
I second the vote for yes on mm

thormj, 2018-02-16:
Third.  Things stored should always be stored in mm.  If we want the user to be able to enter them in inches, then we use the config dialog to calculate it and then store it in mm so we're consistent.