MaslowCNC/GroundControl #632: Limit the Z Axis travel of the G38.2 code

blurfl, 2018-02-16:
I noticed a misunderstanding of the G38.2 code in the recent Z-Axis changes. The 'Touch Zero' button is currently disabled, but the logic to activate it is present. PR #631 addresses this.

The G38.2 code is designed to accept a negative Z parameter which defines the limit to which the Z axis will probe before cancelling the action. It is important that the G38.2 command not drive the Z axis beyond the physical limits of the machine and the Z parameter helps prevent this.
The intended process is for the user to manually move the Z axis near to the surface and set that starting point as zero, then issue the G38.2 command to probe for the surface. If the probe does not touch before reaching the limit defined by the negative Z parameter, the command fails. If the probe touches the surface, that location becomes the Z-axis zero point.

thormj, 2018-02-16:
What happens when the "command fails"?
I'm thinking the default probe distance should be relatively short (say 1/2") and relative -- if it probes and doesn't find, it stays at the bottom and says "no touch detected..." and the user could try again if it just isn't deep enough yet (or he could go up using the relative mode, close, then reposition the sled, and try again)

thormj, 2018-02-16:
Would it be "safe" to assume the "z-before probe" is a good traverse depth if calibration isn't done yet?
Or better to disable the absolute buttons until calibration is done?

blurfl, 2018-02-16:
When the "command fails", Z travel stops and a message is printed to the textConsole:
 error: probe did not connect
 program stopped
 z axis not set

 The Z parameter is required, if it is not included the code fails as above. I always use 0.2" for a Z parameter. This is a short travel that I can be sure will not bottom out and strip the mechanism if the touch probe malfunctions (e.g. clip slips off).
 The user can 'try again' by setting the present location as the z-zero and re-issuing the G38.2 command. I don't think changing to relative mode is necessary. Using absolute values means it will work from any starting position so long as the zero is within the Z-parameter of correct.

 "z-before probe" is not necessarily a good traverse depth, but I don't see a need to disable any buttons.

mrfugu, 2018-04-05:
along these lines, if it isn't yet included, I would encourage some sort of sanity check on all Z-Axis movements to be 'reasonable' (maybe just outside the maximum travel of the default router) or immediately fail. 

I know there are folks who are building more complex (than the default) Z-axis assemblies, so maybe access to modifying these hard limits would be good to add to the advanced option settings. (which themselves could use some more sub-categorization)

blurfl, 2018-04-05:
The zero point gets moved frequently, and is relative to the bit being used. How to determine what 'reasonable' movements are from there?

mrfugu, 2018-04-05:
i mean in terms of maximum outside boundaries, to prevent damage, etc. The total travel of the Default router is less than ~3-4", assuming an end mill of any length, and the work surface, usable distance is even less, and the Z-axis shouldn't ever be allowed to move in any one direction greater than that. 

As it is, i assume I *could* enter 500mm into the Z-axis and it'll try to do it, and fail. If not, please ignore, but GC is still a little rough around the edges, and I figure this might be some help.

blurfl, 2018-04-05:
Now I understand. A reality check in GC's Z-Axis panel distance setting for the 'Raise/Lower' value used in that dialog. That's certainly workable, what value would you suggest? Half an inch, maybe? Longer travel is just another click or two.

davidelang, 2018-04-05:
On Thu, 5 Apr 2018, mrfugu wrote:

> along these lines, if it isn't yet included, I would encourage some sort of 
> sanity check on all Z-Axis movements to be 'reasonable' (maybe just outside 
> the maximum travel of the default router) or immediately fail.

keep in mind that there is no way of knowing where you are starting, so any Z 
movement is a risk, you may already be at the physical limit and the system 
doesn't know where that is.