MaslowCNC/GroundControl #633: Should be able to redraw the canvas without forcing a g-code reload

thormj, 2018-02-16:
Right now, several tools (including my background image thing) want to re-draw the canvas.  It seems the way to do that is to set the GCodeFilename to "" and then reset it (but that would reload the file (without the user telling us to do so), and would definitely not be safe to do while the machine is executing g-code...

So, there is a weird failure that if you set the background image thing to "auto-load from a directory" and you're cutting and you snap a picture, bad juju could happen...

blurfl, 2018-02-16:
What are the tools re-draw the canvas?

blurfl, 2018-02-16:
Would calling gcodecanvas.clearGcode() followed by gcodecanvas.reloadGcode() do what you're after?

thormj, 2018-02-16:
That's what happens now... but doesn't that clear and reload the gcode buffer that is being fed to Maslow?
Or is that "g-code file" someplace else?

blurfl, 2018-02-16:
You're right, that's just what it does. I lost sight of your goal. If I understand, then, what's wanted is a way to draw an overlay (underlay?) without affecting the rest of the gcodecanvas. Does '[canvas.after'](http://robertour.com/2013/07/19/10-things-you-should-know-about-the-kivy-canvas/) look useful? The gcodeCanvas might call the background draw code after it draws?

thormj, 2018-02-16:
Ooh... that might work.  I'll have to play with it and see what happens.
Otherwise, it looks like we should have a "g-code buffer" that doesn't get touched unless asked, and the g-code canvas / maslow-send-file/etc should all read from _that buffer_ and not from the file.

I'll take this guy on; if we do the 2nd  one, we can probably also knock out issue 628 (since the code _as it is on disk_ will be presented to the user (so it will have all comments and stuff).

blurfl, 2018-02-16:
The gcodes sent to the firmware are necessarily limited in length, so stripping the comments and formatting the lines is important.

BarbourSmith, 2018-02-16:
>  it looks like we should have a "g-code buffer" that doesn't get touched unless asked

I think we do! data.gcode is the text of the file after being run through some regular expressions to remove comments and the like. Right now we are touching it by changing the file name, and I agree that is bad practice and we should not be re-reading the file from disk without asking.

blurfl, 2018-02-17:
I'm having trouble following along with this, it doesn't seem to be needed.

Once data.gcode is built, why would we need to rebuild it unless we are re-reading the file?
 - Doesn't data.gcode have everything the machine and the display need?
 - Doesn't updateGcode() "parse the gcode commands and calls the appropriate drawing function for the specified command"?   and 
 - Doesn't clearGcode() "clear the lines and arcs corresponding to gcode commands from the canvas"?
These all seem to operate without needing to reload the file.

 If the issue is the 'centerCanvasAndReloadGcode()' function which centers the canvas and then reloads the file, how about providing a call to center the canvas, clear it and re-draw it? 'reloadGcode' and 'centerCanvasAndReloadGcode()' don't seem to get called except in what seems like appropriate places:
 - at startup
 - if the 'Open File' button or the 'Update Gcode' buttons are picked

If the issue is the 'Update Gcode' button, isn't the purpose of that to re-read the file as it does? That's what I use it for, and pretty often...

thormj, 2018-02-17:
After implementing it and thinking about it... it really seems like I'm the only one who needs it (and if I can figure out how to have a "canvas under the gcode-lines" I won't really need it) currently.

I needed a way to redraw the g-code after loading the the image.  I was centering and reloading the g-code, but that was a bad idea.  In the Pull Request #636, I implemented this and fixed a few other things (now you can see the original g-code as well as the shortened version, and it was reloading the g-code if you shifted it).

So... I guess the "issue" was caused by me, but I'd still like to get PR #636 and then PR #613 in place (if only because I'm finding the "rebasing" to a pain that always needs to be fixed (maybe I'm not  doing it right)...

thormj, 2018-02-20:
Merged into 1 commit since the CanvasImage was getting unwieldy.

closed 2018-02-20 by thormj