MaslowCNC/GroundControl #638: GC assumes it can reset the target board.

blurfl, 2018-02-19:
GC assumes that when it makes a connection to the board running the firmware, the board will do a hardware reset. It uses a "change parity/close/open/close/change parity/open" reset sequence in serialPortThread::getMessage to effect this.
 Not all boards react to this signaling, Teensy3.5/3.6 for instance. Those cannot be reset using the USB connection, so on connection they continue to run, reporting current location and waiting for commands.
 After connecting GC is waiting for 'ok' from the target board. One way to get a response like that would be to send a command immediately on connecting which would cause the firmware to respond 'ok'. Currently the most benign choice looks like sending '? \r\n' as the firmware ignores '? \r\n' but acks it.
 I think there is talk of changing the link between GC and firmware to have GC send '? \r\n' whenever it wants to know the present location, as grbl does. Rather than writing that major change, I'd like to have the serialPortThread send '? \r\n' after doing the Mega'reset sequence. As a bonus, It looks to me like this could in fact replace that dance, but that needs some verification through testing by others 🙂.

thormj, 2018-02-21:
I wouldn't use the ? command; grbl (I think correctly) doesn't respond with OK when a stat inquiry is commanded, just the `<reportOfStuffHere>`.  Now GRBL does response with OK if a blank crlf is sent (if it's not doing anything else), so sending an empty line could work.

If we really want to soft-reset things over the serial port, we should send (and write the firmware to handle) a `ctrl-X`.  In the grbl world, though, that means that all positions (X,Y,Z) are invalidated, spindle/coolant is shut off, then it waits just a bit and reboots itself.  I've not had to "reload chain values" from Maslow when stopping/starting GroundControl, so I'm not sure I'd like to do that...

I suppose the  "compatible lesser extreme" would be to send a !, which will report an OK.

blurfl, 2018-02-21:
There was talk of changing GC to use the '?' query to ascertain the machine's position, ala grbl. Once we have that in place, this issue will go away, as GC will no doubt enter into that exchange on connection.
 Until then, the '?' as it is already implemented is a do-nothing placeholder with no impact on the firmware other than to generate an 'ok' response. The '!' command changes the machine's state, that's why I hesitated to suggest it.
 Using the '?' in this way seems a safer choice to me, as it requires no change to the firmware, and only a one-line change to GC.

thormj, 2018-02-21:
Oh...ok.  So it's not that you're wanting a stop/reset, it's that you're needing the OK to tell GC that it's up and running and not just an empty serial port.  Doing '?' sounds like a better idea then (and when the FW changes, we'll just update GC to look for the stats line instead of "OK").