MaslowCNC/GroundControl #640: The way the calibration process is written is clunky

BarbourSmith, 2018-02-20:
The calibration process was written as a quick and experimental idea of how to measure the machine and it hasn't been updated although it has grown in complexity quite a bit. 

I've been spending quite a bit of time maintaining it over the last few weeks and I think that it's time we re-imagined it from the ground up to reduce the complexity.

I think we need to get rid of the carousel entirely. The carousel is a very linear structure and we have a somewhat tree like process where there are multiple options for how to progress through the process depending on things like if the chains loop over the sprockets to the left or right and if the kinematics type is triangular or quadrilateral.

I propose that we make a catch all widget which gives us the frame and the forwards and back buttons, then have that widget keep a list of steps to move through. Each step is a widget which can be loaded inside the frame letting us easily add or remove steps or do them in a different order.

thormj, 2018-02-21:
I like that idea.  I'd actually like to re-imagine it a bit so that it becomes a "machine setup" process and code be able to add/remove pages depending on what was enabled; eg

a "Z-Axis" module that would ask if manual or powered z... if powered Z, it could ask for the pitch and if you wanted use a Z-touch probe - which would should show how to hook it up and ask if the touch probe was a puck or fixed to the back of the board.

a BackgroundImage module - which would go over the setup for that (how to measure the markers and if you want to autodetect them a color picker for them).

A piece that lets you pick the shape of mechanics and then presents the correct set of calibration screens for that shape (and won't ask you for "distance between links" if you chose "triangular"

etc...  That way if a new bit is added (eg "super-polka-dot-calibration" or "spindle speed control"), it can get "baked into" the machine setup screens.

---
As far as /how/ to do this, I think there would be a "Features" list with buttons "set this up" Y/N, and (it would be nice) a way to indicate "New/Improved" to show that it should be re-done (maybe each module returns a "revision number" as well and if the "revision number" from the module is "more than the ini file" it flags it as "New."

Then you can have a Navigation <<>> bar on top of a screen that is full-sized and each of the modules could be called as a pop-up with which will call the Navigation bar's "Next/OK" or "CANCEL" methods (and if someone clicks the navi-bar, the bar closes and goes to the next step.

BarbourSmith, 2018-02-21:
I like that idea! 

That's very similar to how I am trying to implement it. Basically there is a list of steps and steps can be re-ordered, added or deleted at any point. Each step has a function which runs when the step is entered, and a function which runs when it finishes. When it completes it tells the frame to load the next step.