MaslowCNC/GroundControl #646: .jpg vs .JPG (Linux issue)

GeroBH, 2018-02-21:
An old error has sneaked back in, noticed during calibration on Linux, that for example the image:
GroundControl-master/groundcontrol.kv:691:                    source: "./Documentation/Calibrate Machine Dimensions/Test Shape Preview Triangular.JPG" is not loaded because the image has the extension .jpg

Kindly allow me to try and clean this up myself.
Checking the code I found 11 .jpg and 4 .JPG, that seems to be a clear vote for .jpg.
Unless there is a specific reason to keep capital letters, I would like to attempt a clean-up on the code and image names.

Reference .JPG:
GroundControl-master/README.md:12:[image]
GroundControl-master/groundcontrol.kv:691:                    source: "./Documentation/Calibrate Machine Dimensions/Test Shape Preview Triangular.JPG"
GroundControl-master/groundcontrol.kv:1131:                        source: "./Documentation/Calibrate Machine Dimensions/Quadrilateral Kinematics.JPG"
GroundControl-master/groundcontrol.kv:1133:                        source: "./Documentation/Calibrate Machine Dimensions/Triangular Kinematics.JPG"

Reference .jpg:
GroundControl-master/groundcontrol.kv:761:                    source: "./Documentation/Calibrate Machine Dimensions/Chain On Left Sprocket.jpg"
GroundControl-master/groundcontrol.kv:763:                    source: "./Documentation/Calibrate Machine Dimensions/Chain Between Sprockets.jpg"
GroundControl-master/groundcontrol.kv:765:                    source: "./Documentation/Calibrate Machine Dimensions/Chain On Right Sprocket.jpg"
GroundControl-master/groundcontrol.kv:810:                    source: "./Documentation/Calibrate Machine Dimensions/Set Z Height.jpg"
GroundControl-master/groundcontrol.kv:844:                source: "./Documentation/Calibrate Machine Dimensions/Sprocket at 12-00.jpg"
GroundControl-master/groundcontrol.kv:922:                    source: "./Documentation/Calibrate Machine Dimensions/Ready To Calibrate Right Sprocket.jpg"
GroundControl-master/groundcontrol.kv:924:                    source: "./Documentation/Calibrate Machine Dimensions/Sled Attached.jpg"
GroundControl-master/groundcontrol.kv:1033:                    source: "./Documentation/Calibrate Machine Dimensions/Chain Between Sled Mounts.jpg"
GroundControl-master/groundcontrol.kv:1063:                    source: "./Documentation/Calibrate Machine Dimensions/Measure Motor Height.jpg"
GroundControl-master/groundcontrol.kv:1161:                        source: "./Documentation/Calibrate Machine Dimensions/Test Shape Preview.jpg"
GroundControl-master/groundcontrol.kv:1163:                        source: "./Documentation/Calibrate Machine Dimensions/Measure Test Shape Horizontal.jpg"
GroundControl-master/groundcontrol.kv:1165:                        source: "./Documentation/Calibrate Machine Dimensions/Measure Test Shape Vertical.jpg"

reecej, 2018-02-21:
Great find!

BarbourSmith, 2018-02-21:
Excellent catch! I see there is a pull request already! Great work 👍 👍 

I will try to keep more of an eye on it in the future and stick with .jpg

closed 2018-02-21 by GeroBH