MaslowCNC/GroundControl #651: Changes to kinematics type are not pushed to machine?

BarbourSmith, 2018-02-23:
I'm running into an issue which I haven't quite tracked down yet, but I wanted to post it anyway to see if I could get some feedback.

The general issue that I'm seeing is that my new calibration routine is producing junk results. After I finish the calibration process my machine is not even close to accurate...so something is wrong.

I assumed it was something in the new calibration routine which was causing the issue since that's what changed. It almost seemed like my changes weren't getting pushed to the machine so I went to look to see if my settings are getting pushed up at the right time and they all seem to match up except that my machine is in quadrilateral mode while it should be in triangular.

[image]

When I check my settings file itself things look good (triangular):

[image]

I think the issue may have to do with the way the settings are checking to see if the settings on the machine side match up with the settings stored on the arduino side. The function is like this (with my debug lines added):
```

def syncFirmwareKey(firmwareKey, value, data):
    for section in settings:
        for option in settings[section]:
            if 'firmwareKey' in option and option['firmwareKey'] == firmwareKey:
                storedValue = data.config.get(section, option['key'])
                print "stored value is: " + str(storedValue)
                if not isClose(float(storedValue), value):
                    print "value is different and needs to be updated"
                    data.gcode_queue.put("$" + str(firmwareKey) + "=" + str(storedValue))
                else:
                    print "value is the same"
                    break
    return
```

And in the terminal I am seeing:

```
$7=1 (Kinematics Type 1=Quadrilateral, 2=Triangular)

1.0
2
passed on
stored value is: 1
value is the same
```
When in reality the stored value isn't 1 it's a string "Triangular" which is being read as 1...?? this is where I get lost.

Anyone have any thoughts? @krkeegan what am I missing?

krkeegan, 2018-02-23:
That certainly seems wrong.  Let me see if I can see the same thing.

BarbourSmith, 2018-02-23:
When are the `Computed Settings` updated

krkeegan, 2018-02-23:
Hrm, I don't see the same thing.  I hope this isn't a platform distinction.  

`Computed Settings` is updated in configSettingChange which is called anytime a setting is altered.

BarbourSmith, 2018-02-23:
Great! I'm glad you don't see it too, I'm guessing I broke something in a nuanced way. I see where configSettingsChange is called and I will start digging there 👍

krkeegan, 2018-02-23:
So I did this exercise:

1. Set the type to quadrilateral. (Watch the terminal, you should see the $7=1 command sent.)
2. Unplug arduino
3. Set the type to triangular
4. Close GC
5. Plug in arduino
6. Start GC (you should see at some point in the startup Sending: $$ ... Sending: $7=2)

That all works for me.

Similarly if I just change the type in the settings, I see a $7 command sent for every change.

BarbourSmith, 2018-02-23:
Yeah, everything works exactly as it should when I just go into settings and switch from one to the other, it just doesn't seem to work when I change the setting remotely using the config.set from somewhere else. 

It feels like a kivy binding bug to me...or maybe my implementation is just wrong 🙄

krkeegan, 2018-02-23:
You want to push your commit and I can see if I can spot the problem?

BarbourSmith, 2018-02-23:
I'm seeing the issue in #652 but I'm pretty sure it's in my side of the code so if nothing jumps out immediately don't worry about it.

I added some more debug lines to print every time the binding gets called and it's working right so I can't blame a kivy bug anymore 😢

BarbourSmith, 2018-02-23:
I think I know what is going on. We've run into this bug before where the value reflected in the settings doesn't update when it is set externally using the .set() method. 

If a value is set with .set() and then you click the settings button to check that it is correct it will show the wrong value until GC is restarted. It is a Kivy thing that we decided was very tough to fix. 

I'm not sure that's the whole issue, but that explains some of what I'm seeing.

krkeegan, 2018-02-23:
Seriously?  Well that could be an issue then.  I relied pretty heavily on the fact that set() would work properly.  This could be leading to a lot of issues then.  How do we ensure that the values are correct without a restart?

krkeegan, 2018-02-23:
Hmm, ok I see the bug even without your code, when I am working in the calibration routine.

BarbourSmith, 2018-02-23:
I think everything works ok behind the scenes, its just the displayed values which aren't updated. So .set() works right and all the bindings are triggered, just the value that we are seeing when we open the setting panel isn't updated

BarbourSmith, 2018-02-23:
I think those values are only loaded once when GC launches

krkeegan, 2018-02-23:
Hmm, if I add a debug at the top of configSettingChange() I don't see it getting called when I make changes in the calibration screen, hopefully this is just a binding issue.

krkeegan, 2018-02-23:
I am working on this.  I think I must have done something wrong.

BarbourSmith, 2018-02-23:
It's been working like a charm until I re-did the calibration process so I'm pretty sure I broke it 😀 

I'm digging too on my end

BarbourSmith, 2018-02-23:
I am seeing something very strange where sometimes as I am going through the calibration process I can see all of the callbacks get triggered as I change settings in each step...and sometimes nothing is triggered...I'm not sure what the deciding factor is yet

BarbourSmith, 2018-02-23:
So I think I've figured out how to make each of the two cases I was seeing happen.

To make the settings not push when changes are made during the calibration process:

1. Delete the .ini file
2. Launch GC
3. Select the COM port -> Watch as $$ is sent and settings are synced
4. Open the calibration process and nothing happens as each step is done

To make the settings push with each change in the calibration process:

1. Delete the .ini file
2. Launch GC
3. Select the COM port -> Watch as $$ is sent and settings are synced
4. Open the settings and change ANY number
4. Open the calibration process and watch as the settings are pushed with each change

krkeegan, 2018-02-23:
Oooh interesting.

krkeegan, 2018-02-23:
I can do you one better.  You can skip the delete ini and the change any number.

Simply just opening the settings dialog is doing it.  So I think something in the settings panel needs to be initialized.

BarbourSmith, 2018-02-23:
I think the issue is that the build_settings() function is not run until the settings window is opened 👍 👍 

Is it safe for me to force call it from somewhere else during startup?

krkeegan, 2018-02-23:
I tried moving it after:
config.remove_callback(self.computeSettings)

While I see the callback called, it isn't sending the command to the arduino.  

Progress.

BarbourSmith, 2018-02-23:
moving `self.config.add_callback(self.configSettingChange)` up to line 105 in `main.py` so the callback is setup outside of the settings window being opened seems to be working for me

krkeegan, 2018-02-23:
Its better.  But I still see some odditity.

If you go to page 9 of the settings and  pick quadrilateral, you can see the message sent the first time, but go back an pick triangular or anything else and nothing gets sent.

Seems like some error in the callback itself now.

krkeegan, 2018-02-23:
No it seems to be working for me now.  Not sure what that was.

We do need to go through and remove most of the calls the pushSettings as unnecessary, that may have caused my issue.

Also most of the calls to write() are also unnecessary.  As this is handled in configSettingChange()

BarbourSmith, 2018-02-23:
Are you using the new version of the calibration or the old version? 

I'm seeing everything working OK on the new version if I go past choosing quadrilateral and then go back and pick a different option I see `Sending: $7=2` every time

BarbourSmith, 2018-02-23:
I've updated #652 with the version I'm testing right now

krkeegan, 2018-02-24:
Yeah, it all seems to be working for me.  I think the issue was some hiccup with the call to pushSettings().  I sent you a PR to #652 that removes the unnecessary calls to pushSettings and write.  Test it, but it all works as expected on my end.

BarbourSmith, 2018-02-24:
Fantastic! Thanks for figuring that one out with me. I thought I was going crazy all morning. 

I'm going to close this

closed 2018-02-24 by BarbourSmith

krkeegan, 2018-02-24:
Agree, the python stuff particularly with kivy gets a bit tough for me as I don't really understand all the nuances of the framework (as demonstrated).  

I strongly suspect this issue may be subtly causing a lot of issues for people.

madgrizzle, 2018-02-26:
Might explain a lot of strange behavior.. such as sleds falling to the floor during calibration, etc.