MaslowCNC/GroundControl #655: Feature Request: Toggle dashed traverse lines

bazzacad, 2018-02-26:
Sorry if this setting is already available, I didn't see it.
I'd like a way to toggle the dashed traverse lines on & off.
When I first open a gcode file, I just want to check if the parts/paths look correct.
The traverse lines make things cluttered & hard to see.
But when I start the cut, I do want to see them, so I know where the router is moving...

BarbourSmith, 2018-02-26:
Yes! I think this is an excellent idea.

Help us better understand how you want to see the feature implemented. Would there be an option in the settings panel or would there be a button on the front page somewhere to make it easy to toggle...What about in the spiner menu that comes up when you press and hold? That way it wouldn't clutter the UI up too much more, but it also wouldn't be burred in the settings.

bazzacad, 2018-02-26:
Actually I totally forgot about the spinner menu, thanks for the reminder. ;-)
Ya the menu would be a good place, but also burring it in the settings (but show a keyboard short-cut) would be good too. Maybe something like CTRL+T (for Traverse)... Just a thought...

BarbourSmith, 2018-02-26:
I like the keyboard shortcut option...we should work on a widget for making it easy to add keyboard shortcuts...but that's a separate issue.

Sounds like a plan!

I probably won't be able to get to this by Wednesday for the release, but I'm on it!

blurfl, 2018-02-26:
How about a mode to step through the traverses highlighting them one at a time, to reveal the order?

BarbourSmith, 2018-02-27:
I started working on that but I got bogged down trying to make an arrow 🙄 ... which I am now embarrassed to admit