MaslowCNC/GroundControl #656: Red & Green start & stop circles

bazzacad, 2018-02-26:
I'm not sure if this is a bug or a design intent, but the red & green start & stop circles seem to be opposite of what I'd think they should be. If you think from the cutting point of view, the start of the cut is a red circle & the stop of the cut is the green circle, I'd think they'd want to be the other way around. But if you're thinking from the traverse point of view, then they're correct, green start of traverse, red stop. I tend to think form the cutting point of view. 

[image]
of traverse.

BarbourSmith, 2018-02-27:
Woah! I was thinking of them as red being dangerous because the bit is lowering, and green being safe because it's lifting up

blurfl, 2018-02-27:
"red being dangerous because the bit is lowering, and green being safe because it's lifting up" That's the way I looked aty it... Also red stands out more, and that's where the cutting begins.

bazzacad, 2018-02-27:
OK, never mind ;-)

BarbourSmith, 2018-02-27:
Let's leave it as is just because it doesn't seem like a pressing issue, but I'm glad you brought up the idea of red and green possibly indicating start and stop. That makes a lot of sense.

krkeegan, 2018-02-27:
I believe that the simulator in Fusion360 uses the red as plunging and green as retracting as well.

closed 2018-02-27 by BarbourSmith