MaslowCNC/GroundControl #659: Unable to find position when calibrating

BarbourSmith, 2018-02-27:
There are a number of steps during the calibration process where the chains are not of valid lengths for the machine to find it's position.

This is normal, but the popup shows up anyway saying that it couldn't find a valid position which will confuse everyone.

We want the popup to not show up. Do we suppress it on the GC side, or make the forward kinematics not run on the firmware side? I think doing it in GC makes the most sense, but we want to be careful not to suppress it when we don't mean to.

krkeegan, 2018-02-27:
I have thought about this as well.  And I think I agree with your conclusion, GC should really be the one ignoring it.  The error is correct, we just don't want to bug the user with the details during a specific process.

Alternatively, we are talking about putting the machine in some state when we enter the calibration process and clearing that state when we exit.

blurfl, 2018-02-27:
>putting the machine in some state when we enter the calibration process and clearing that state when we exit.

Not sure if this is germane, but I regularly go into the calibration sequence to access the sprocket zeroing tool, then leave to use the manual chain calibration. Watch for unconventional paths through the maze...

krkeegan, 2018-02-27:
Yeah, think ignoring it in GC is the way to go.  With the only hiccup of how do we tell a user that the machine is not calibrated at the end of the process?  Such as they leave early, or somehow we really messed something up in the process.  I suppose leaving early can be captured by the general warning about leaving early.

davidelang, 2018-02-27:
the forward kinematics code should not run until quite late in the process (when 
we have the chains fed out from both motors and connected to the sled)

Everything prior to that should be done with single-motor drive commands 
(distance and/or power).

what is causing the routine to be run?

krkeegan, 2018-02-28:
Forward kinematics is called whenever any setting that affects kinematics is changed.  There is no additional call to $K to rerun kinematics anymore.  This makes changing settings from a strict command line more straightforward.

BarbourSmith, 2018-02-28:
> With the only hiccup of how do we tell a user that the machine is not calibrated at the end of the process? Such as they leave early, or somehow we really messed something up in the process.

If they leave early it makes sense to me that they would get a warning about the chain lengths being off. In the case @blurfl pointed out about using the calibration to move the sprockets to zero he would get the warning, but would know to ignore it because it was intentional.

commit a8ffe035 (2018-02-28): Add a flag to hide messages about chain lengths being invalid during calibration

Also some general cleanup

Fixes #659
PR #660 (2018-02-28): Add a flag to hide messages about chain lengths being invalid during calibration

Also some general cleanup

Fixes #659
closed 2018-02-28 by BarbourSmith