MaslowCNC/GroundControl #668: GroundControl Error message

robertkni, 2018-03-16:
i'm not sure what's going on but am getting this message and I cannot calibrate or move motors or anything.  Anyone have a solution?

[image]

BarbourSmith, 2018-03-16:
Hmmm that message means that the firmware has not yet gotten information from ground control about what the machine dimensions are.

What version of Ground Control and what version of the firmware are you using?

robertkni, 2018-03-16:
ground Control v1.09
Newest Firmware from 2 days ago: https://github.com/MaslowCNC/Firmware/releases/tag/v1.09

BarbourSmith, 2018-03-16:
@krkeegan do you have any thoughts on what could be causing this?

We could probably make the issue go away by forcing an GC to push the settings with a $$ but that seems like just masking the issue and it would probably have strange results down the road

robertkni, 2018-03-17:
I restarted computer, opened the arduino file, made sure the board and port was correct, then uploaded.  I then opened GC and this is the message i got right away.

[image]

davidelang, 2018-03-17:
try wiping the flash on the arduino and re-load the firmware.

robertkni, 2018-03-17:
Just started from scratch and still getting same error.  I completely deleted Arduino, Ground Control & files, restarted computer.  Reinstalled everything and went through entire process again.  Same error message.

robertkni, 2018-03-17:
@davidelang

blurfl, 2018-03-17:
In GC, on then Actions page, in the lower right corner, the Advanced button brings up a list, the action you want is 'Wipe EEPROM' that list.