MaslowCNC/GroundControl #670: Need button to ignore 38.2 probe GCODE

mnelson78, 2018-03-24:
I had great success with generating G code in easel and then cutting in ground control. I did have to manually edit the file though as the code had the 38.2 probe command and I don’t have that set up on my maslow. It would be nice to have a feature to ignore this G code command it someone isn’t using probing without having to edit the g code file.

blurfl, 2018-03-24:
I use easel to generate code, and have never had it put g38.2 into a file. What settings were you using in easel?

blurfl, 2018-03-24:
I see it, here's what's going on.
 Easel offers to work with three different machine configurations - X-carve,  Shapoko, or Carvey. Carvey is their most 'user-friendly' machine, sold with much built-in automation to do configuration for the user. Part of that is the automatic z-axis zeroing. When configured for Carvey, Easel adds the G38.2 code, and it also adds other gcodes (G4, G10, G54 - dwell time, and alternate coordinate system) that aren't (yet?) implemented by the Maslow firmware. These are associated with the automation added to make the Carvey a hands-off device.
 Neither of the others (X-carve,  Shapoko) add these codes, so choosing either one of those will produce an .nc file that 'just works' with the Maslow.

mnelson78, 2018-03-24:
Awesome!  Thank you!  

No need to add this to ground control then  :-)

closed 2018-03-24 by mnelson78