MaslowCNC/GroundControl #672: gcodeCanvas fails to render full circles (but the firmware cuts correctly)

blurfl, 2018-03-28:
If the starting XY and ending XY of a G02 or G03 code are identical, the circle will not be rendered on the gcodeCanvas. When that same line is sent to the firmware, it will be correctly processed.
 Here is a short sequence in which the G03 code will render a circle because the endpoint XY is different from the starting point:
```
G20 G90 G0 X0.0 Y0.0  (move to starting point)
(in the next line Y differs from starting point so it renders)
G3 X.0 Y0.001 I2.0 J0.0 F100 
```
Here are three valid G03 codes all cut correctly in the firmware but do not render:
```
G20 G90 G0 X0.0 Y0.0 (move to starting point)
(X,Y same start and end, does not render)
G3 X0.0 Y0.0 I2.0 J0.0 F100 
( G03  with I, J - X, Y default to zero, does not render)
G03 I2.0 J0.0 F100 
( G03 with I, - J, X, Y default to zero, does not render)
G03 I2.0 F100 
```

BarbourSmith, 2018-03-28:
Good catch! This corner case has been dogging us since day. 

Thanks for the clear test case. Being able to consistently replicate the issue is a huge first step to getting it fixed.

blurfl, 2018-03-28:
Can you hint at where I should look to work on it?

BarbourSmith, 2018-03-28:
I think the issue has to do with how the inverse tangent function works in line 308 and 309 in the file UIElements/gcodecanvas.py...If I remember right the actual result of an inverse tangent where both the points are the same is both 0 and 360 so the returned value is not determined.

Apparently the arduino folks went with one option and python went with the other 😸 

Maybe a special case to catch it? Although special cases always seem to get me in trouble

blurfl, 2018-03-28:
I'm right ther, looking at it. Looks promising, too. Thanks for the advice!

BarbourSmith, 2018-03-28:
Sorry it took me 7 hours 😬

closed 2018-04-13 by blurfl