MaslowCNC/GroundControl #682: Auto-calibration might reality-check user input and calculated values

blurfl, 2018-04-01:
Depending on program flow, the values of various settings, and user confusion, values recorded in groundcontrol.ini can be quite wrong. Clicking 'Measure' without extending the chain in the panel which measures the distance between motors puts -12.7 into motorspacingx (-0.5 inch, the two links that would need to be subtracted if the chain was on the right sprocket). This issue could serve as a collecting place for issues like this with discussion of reasonable values and possible remedies.

blurfl, 2018-04-01:
Here's is one to start...

The proposed value for a measurement could be shown to the user before being committed to the groundcontrol.ini file, with the option to 'accept' or 're-do', or maybe to leave the value unchanged. Even better, also show the previous value (forgive the lame markdown formatting...)

Prompt what is going to happen if you accept this  | value 
------------------------------------------------ | --------------
the spacing between the motors was previously | 3006.794mm
 but this measurement will change that to | 2994.09mm
[accept] [re-do] [leave unchanged]

blurfl, 2018-04-01:
The value proposed to fill 'motorspacingx' could be compared to the 'bedwidth' value in groundcontrol.ini, and the user prompted if the new value is less than 'bedwidth' plus some assumed reasonable amount. The prompt might ask the user if the motors are wider than the bedwidth, a reasonable assumption but one that should be checked - a top-beam frame could have the motors mounted anywhere along that beam...

GeroBH, 2018-04-01:
A calibration walk-through with sanity check on values sounds great!

We should already know the rotation radius of triangular mounts by choosing one option:
(This assumes that you have all links needed to connect the sled, attached to the chain during calibration.) 
- ring kit latest shipment. rr = ?
- pillageTHENburn kit. rr = ?
- dlang 5 Inch kit. rr = 5 Inch
- dlang 7 Inch kit. rr = 7 Inch
- user input option for diy freedom