MaslowCNC/GroundControl #683: Topfeeding - "Ready To Calibrate.jpg" Image Clarification

madgrizzle, 2018-04-02:
Just wanted to point out that it might be a source of confusion, that the topfeeding image of "Ready To Calibrate.jpg" is from the right motor and is in the same configuration as the bottomfeeding image from the left motor.  Can we perhaps show two images, either top feed left / top feed right or bottom feed left / bottom feed right, as appropriate?  I'm just looking for points of confusion for new people.

BarbourSmith, 2018-04-02:
Great point. I'm about to start testing calibration issues and I will look into making this two pictures as part of that

BarbourSmith, 2018-04-02:
This picture should actually already update based on if the machine is set to be top feeding or bottom feeding.

If the machine is in the bottom feeding configuration, the picture on the left is shown where the left chain feeds off the left side of the sprocket. 

If the machine is in the top feeding configuration, the right motor is shown because the chain is still extended on the left motor:

[image]

madgrizzle, 2018-04-02:
Understood, but I think two images would really help clarify.. especially in top feed configuration you show the chain on the left sprocket dangling down (people have been confused as to whether to reset the left chain to 12 o'clock or leave it dangling)  It could be one actual image file, just two pictures merged into one jpeg.

BarbourSmith, 2018-04-02:
Great suggestion. Making two images in one should be an easy and quick solution.

davidelang, 2018-04-02:
seeing the confusion here, I'm thinking that it may be simpler to have 
step-by-step pictures and after measuring the motor distance, go ahead and reset 
the sprocket to 12oclock and re-hang the chain no matter which direction you are 
going.

while it saves a little time to not re-hang the chain and just retract it, the 
confusion that's caused seems like it eats up more time overall.

BarbourSmith, 2018-04-02:
That's a really good point @davidelang

madgrizzle, 2018-04-16:
It seems like it would be possible for GC to get the sprocket to 12 o'clock..  Is this correct?  I don't know the best way of doing it, but I think it would be possible.

BarbourSmith, 2018-04-16:
You are right! We know the number of sprocket teeth and we know the number of times it has rotated since it was last at 12 o'clock...very cool thought

davidelang, 2018-04-18:
On Mon, 16 Apr 2018, madgrizzle wrote:

> It seems like it would be possible for GC to get the sprocket to 12 o'clock.. 
> Is this correct?  I don't know the best way of doing it, but I think it would 
> be possible.

When you are in the calibration step, it gives you tools to rotate it.

Unfortunantly, we don't have a way of knowing the absolute position of the 
sprocket, only it's position relative to where we started, so once we loose 
track of where we are, we need outside info (i.e. the user) to get a starting 
point.

the "sprocket pin at 12 o'clock" is not anything magic, it's just a good, 
convienient thing to use.

The requirement is

1. The position must be repeatable (the next time you need to reset the chain 
length, you must be able to get to the same point)

2. The position must be symmetrical on the two motors. If you set both motors 
the same way, attach the chain the same way, and then feed the chains out until 
they meet, they MUST meet in the exact centerline of the machine.

The software is calculating the distance from this 12 o'clock position to keep 
track of how much chain it has fed out, but if we were to always start with 
three (and a half) links past this point instead of half a link past this point, 
the added links would end up being part of the 'rotation radius' number (the 
same thing if you managed to start the chain 3.5 links before this position)