MaslowCNC/GroundControl #693: 1 Deg CCW button does not move motor

golfecholima, 2018-04-10:
1 Deg CCW button does not move motor. Motor does make noise/groan a bit, but does not move.

blurfl, 2018-04-10:
1 degree of rotation isn't really visible. Try giving 5 clicks, and wait for the 'grumbling' to finish with each one. In my experience it really does move, but it's hard to tell. Using a string level on a short piece of coat-hanger wire is a way to check.

[image]

golfecholima, 2018-04-10:
Right, I know that anything less than 5 deg requires a few clicks to see any change. This issue report takes that into account. Nifty level trick 😄 !

blurfl, 2018-04-10:
So do both motors fail to move in response to the 1 degree CCW button? Is it just the first click that fails, or all clicks? Is it possible that this is the result of gear-lash in the gearbox, effective on any direction change? Does it matter if there is weight on the sprocket (is the chain in place)?

golfecholima, 2018-04-10:
It was present on both left and right motors under no load. All clicks fail. Haven't tested with chain or sled installed.

blurfl, 2018-04-10:
Has any of the calibration been done before, or is this a clean slate first calibration?

blurfl, 2018-04-10:
Sorry to be peppering you with so many questions. 
The 1 degree button asks the Mega to move the chain 0.17639mm. It's pretty hard to see an increment so small. That's why I asked about multiple clicks.
 I looked into the code, and it really doesn't matter whether any calibration has been done, the Arduino will use the default values. Sorry I headed down a blind alley there.
 To test further, I removed a sprocket and clamped a 6" straightedge to the motor shaft. On my motors, clicking the 1 degree CCW button moves the distant end of the straight edge an appreciable distance, whether it is the first movement since the motor was powered or after a direction change. At the motor shaft end, though, 1 degree of movement isn't at all easy to see. 0.1 degree of movement is visible at the far end of the straightedge, but only barely.
 If it is one motor having this problem, it might be some problem in the gear train, but I think you said it was both motors. Could you try the straightedge trick?