MaslowCNC/GroundControl #694: Possible error in readMotorSpacing for bottom chain configuration

madgrizzle, 2018-04-10:
This is a snippet of the code to read the motor spacing.  I don't know exactly how the bottom feed is handled in the software, but here it looks like one part is to change the distance between motors to a negative value (multiplies by -1).  The software subtracts off the extra two links for the part that wraps around the sprocket to decrease the measured value for top feed configurations, but the code here for bottom feeding actually adds two links (makes it more negative).  Like I said, I don't know how bottom feed is being handled, but this may be an error.  If it's wrong, then the multiplication by -1 should occur after you subtract the two links.

```
    def readMotorSpacing(self, dist):
        if self.data.config.get('Advanced Settings', 'chainOverSprocket') == 'Bottom':
            dist *= -1

        dist = dist - 2*6.35                                #subtract off the extra two link
```

davidelang, 2018-04-10:
On Tue, 10 Apr 2018, madgrizzle wrote:

> This is a snippet of the code to read the motor spacing.  I don't know exactly how the bottom feed is handled in the software, but here it looks like one part is to change the distance between [...]

given that the motor measurement is the same (in both cases, it's done over the 
top), this looks like leftover code from when Bar was trying to setup the 
measurement to go across the bottom, and should be removed.

BarbourSmith, 2018-04-10:
Good catch!

I believe that the code sets the machine to Top feed in the on_enter() function of that step, then resets it to what it used to be in the on_exit() step so that code should not ever run, but let's confirm that and remove the code before declaring this issue closed

madgrizzle, 2018-04-10:
I see what you mean.  That's a relief.

BarbourSmith, 2018-04-11:
@madgrizzle do you want to take those lines out and make a pull request?

madgrizzle, 2018-04-11:
@BarbourSmith see PR #697

BarbourSmith, 2018-04-11:
Great work!! 🎉 👍 👍

closed 2018-04-16 by madgrizzle