MaslowCNC/GroundControl #696: Calibration issues on Raspberry Pi 3

cskwrd, 2018-04-11:
I am helping my friend calibrating his new Maslow CNC. We are using a brand new Raspberry Pi 3B (not the newest B+).

We are having troubles completing the calibration step because the CNC seems to become unresponsive when we start. We are considering ourselves lucky if we can get one of the motors to turn any amount at least 1 time during the stage where we are to align a sprocket at 12 o'clock. Occasionally, we will get it to move a second time, afterwards it will not move at all.

We have noticed a couple of things. First, we seem to be able to operate the motors from the main screen without any trouble. Once we attempt to calibrate the CNC machine, all bets are off. The other thing we have noticed is that the CPU is pegged at 100% when navigating any of the menus.

Any ideas?

BarbourSmith, 2018-04-11:
I don't have much experience with Linux or Raspberry Pi, but does this thread seem related? I think he mentions some of the same issues:

https://forums.maslowcnc.com/t/gc-connected-no-motor-movement-in-linux/3114

cskwrd, 2018-04-11:
Yes, actually. I am not sure how I missed it.

This [reply](https://forums.maslowcnc.com/t/gc-connected-no-motor-movement-in-linux/3114/14) describes the behavior we were seeing almost exactly. I say *almost* because I don't recall what would happen if we press the stop button.